keyboardio/Chrysalis #667: Some combinations just don't work

snan, 2021-02-25:
For example I have a key set to \| but with the secondary function to shift to layer #2.

Works great. But I'd rather have /? in that position and with that secondary function. I type slashes and question marks more often than backslashes and bars. And that, it does not like.

I clear everything from the key. Pick a key `/`. Add a secondary function "Layer shift when held." It acts if I had just selected modifer when held, altgr, and it also replaces the / with a capslock.

This is on the Keyboardio version of the Atreus with the Chrysalis app and in key position #44.

Rom1deTroyes, 2021-02-25:
What version of Chrysalis do you use ?
There was some bugs on this area on 0.8.2, that were addressed in the last development snapshot... 
... Can you give a try to this one :
https://github.com/keyboardio/Chrysalis/releases/tag/v0.8.3-snapshot

snan, 2021-02-25:
Ok, that worked. Thank you.

closed 2021-02-25 by snan