keyboardio/Chrysalis #668: Chrysalis doesn't work with Sway WM (Wayland Tiling WM)

Deep-Six, 2021-02-27:
**Describe the bug**
Upon executing the Chrysalis app, there is no output other than : 
The futex facility returned an unexpected error code.
[1]    11842 abort (core dumped)  ./Chrysalis-0.8.3-snapshot.AppImage

**To Reproduce**
Steps to reproduce the behavior:
1. Download the AppImage
2. chmod +x the Chrysalis-0.8.3-snapshot.AppImage
3. ./Chrysalis-0.8.3-snapshot.AppImage

**Expected behavior**
Presumably a graphical window would be launched for interacting with the app.

**Screenshots**
N/A

**Desktop (please complete the following information):**
- Arch Linux kernel 5.10.6-arch1-1-oled
-  - Chrysalis Version: 0.8.3

**Additional context**
Add any other context about the problem here.

Rom1deTroyes, 2021-02-27:
Unfortunately it's seems that we need to wait until [Electron get a full support for Wayland](https://github.com/electron/electron/issues/10915)  :-(

If you want to try replacing the chrome part of Elecrton with chromium-ozone (warning : experimental stuff inside !) see :
 - https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=chromium-wayland-git
 - https://chromium.googlesource.com/chromium/src/+/master/docs/ozone_overview.md

I can't tell if it work, as I use an old plain X11 server, but will have a great interest of the result of such manipulations :-)

Deep-Six, 2021-02-28:
I  believe I am already utilizing this feature, as I start Chrome with these args:
"/usr/bin/google-chrome-stable -enable-features=UseOzonePlatform -ozone-platform=wayland"\

algernon, 2021-03-04:
Looks like Electron 12 has some support for Wayland at least, so I made a quick hack that _might_ solve it. Opened #671 with the changes, can you give it a try? Details on how to do that are in the PR.

PR #671 (2021-03-22): Initial support for Wayland on Linux
This upgrades Electron to v12 (which ships with a Chromium that has initial support for Wayland), and adjusts the AppImage launcher script so that if wayland is detected, it will let Electron know we want to use it.

Fixes #668.
commit a9a563b2 (2021-03-22): Initial support for Wayland on Linux

This upgrades Electron to v12 (which ships with a Chromium that has initial
support for Wayland), and adjusts the AppImage launcher script so that if
wayland is detected, it will let Electron know we want to use it.

Fixes #668.
closed 2021-03-22 by algernon