keyboardio/Chrysalis #672: Missing clear layer?

kujohn, 2021-03-08:
**Describe the bug**
There is no ability to clear a layer.

**Expected behavior**
Ability to clear layer

**Desktop (please complete the following information):**
 - OS: Mac OS X 11.2.2 (20D80)
 - Chrysalis Version: 0.8,.2

necsoiu, 2021-04-03:
Or delete it completely, or change its position in the layer stack.

qayshp, 2021-04-09:
Yea, I just ended up exporting, editing, and importing the json. This would be a helpful utility to be baked in.

TrueFalcon, 2021-05-09:
This was possible in 7.9, but it was removed in the 8.# series. You can still install 7.9 if you want to move layers around easily; you will need one empty layer to use as temp if you don't want to lose anything. Then reinstall the latest Chrysalis after. Be aware that any install or reinstall will wipe the directory where it stores yours mods by default and only put back the latest default keymaps. Copy your mods, which ideally have a custom name, to somewhere else and then put it back after install.  If you just want to clear a layer, it's quickest to just set each key to blank or blocked. 

The 8 series of Chrysalis can only save all your layers at once into one file; there's no way that I know of to extract one layer from that file. 7.9 could manipulate and save/install layers individually. I don't know why this feature was removed, maybe some conflict ... 

I do like being able to save all at once though; then I always immediately backup the file to another dir somewhere.