keyboardio/Chrysalis #673: Unable to select layer shift when held for secondary action

kujohn, 2021-03-08:
**Describe the bug**
When selecting `layer shift when held` as secondary action, it defaults to `modifiers when held` and turns on `altGr`.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Mac OS X 11.2.2 (20D80)
 - Chrysalis Version: 0.8.2

Rom1deTroyes, 2021-03-08:
Can you try with the last [development snapshot](https://github.com/keyboardio/Chrysalis/releases/tag/v0.8.3-snapshot) ?
There is a lot of displaying problems that has been addressed sine 0.8.2 :smile:

tzwilliams, 2021-03-16:
@Rom1deTroyes, I had the same problem using 0.8.2 on Windows 10 (20H2).  I just updated to the 0.8.3 development snapshot and the issue seems be corrected.  Thanks!

Rom1deTroyes, 2021-04-04:
Chrysalis-O.8.3 release is out there, so I think this issue can be closed ?

algernon, 2021-04-04:
Indeed, this should be closed as fixed in 0.8.3. Thanks for the reminder, and for the report too!

closed 2021-04-04 by algernon