keyboardio/Chrysalis #677: Consumers keys (Chrysalis-0.8.3-AppImage)

Rom1deTroyes, 2021-03-30:
**Describe the bug**
The consumer keys are not labelled (you see just their numbers

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Consumer control'
2. See the keys under volume, Media Control and Brightness

**Expected behavior**
Get the great new symbols from last snapshot :-)

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Debian10 Gnu/Linux
 - Chrysalis Version: 0.8.3-AppImage official release and custom build from master

**Additional context**
Add any other context about the problem here.

algernon, 2021-03-30:
Off, yeah, we need to figure something out about this. Thing is, the keycodes you see are the _wrong_ keycodes, hence no label. I think the best course of action would be to scan the keymap, and if we encounter any such keycodes, show a warning and a button to transition to the new keycodes.

Rom1deTroyes, 2021-03-30:
Ok, I get it !
I will make a PR to address this asap :-)

algernon, 2021-03-30:
Oh dear. The Sidebar is wrong too, but your PR is going to fix that part!

So sorry, I overlooked this part. Will have to get a 0.8.4 out quick.

Rom1deTroyes, 2021-03-30:
Yes : the problem was double here : 
- Wrong codes on the side bar (:heavy_check_mark: addressed)
- wrong codes on legacy layers (from long time, here) : we must find a clever way to make users figure out what to do :-)

Rom1deTroyes, 2021-03-30:
Is there a need to verify all the keycodes on the application ?
I mean : review the source code, and check if each keycode match the HID code from either Kaleidoscope or directly from the norm ?
I can take time to do that (it's not a hard thing to do with vim and a macro, just take some time and rigor :microscope:).