keyboardio/Chrysalis #681: Get rid of Shift key

necsoiu, 2021-04-03:
**Is your feature request related to a problem? Please describe.**
It will be very nice and usefull to implement this functionality. 🥇 

[https://configure.zsa.io/ergodox-ez/layouts/9Q74x/latest/config/](url)

[image]

Rom1deTroyes, 2021-04-04:
I think you can achieve this with QuKeys ?
https://kaleidoscope.readthedocs.io/en/latest/plugins/Kaleidoscope-Qukeys.html

Personally, I found this kind of time-based comportment breaks my workflow :
- slowing down UPPERCASE words is a non-sense when you abuse of #define in your code
- my typing speed varies all day long (fatigue, thinking vs kilometric inputs, ...), and the timeouts are somewhat inadequate at a point of the day. 
But maybe it's just a lack of practice around this way :thinking:

algernon, 2021-04-04:
We are working on an AutoShift plugin for Kaleidoscope, and when that's done and merged, we can add support for it in Chrysalis too. While it is kind-of possible to accomplish most of it with Qukeys, a dedicated AutoShift plugin makes it easier to support it in Chrysalis too, so we're going to go down that route.

necsoiu, 2021-04-05:
Thank you. It will be very nice to have this. 

Also in time it will be very nice too if you could implement Macros, Tap dance, etc. in Chrisalys. Not all of us have inclinations towards programming, and even if we have, often we do not have the time to make small changes, especially since many times these changes are only temporary.
In the meantime, for the use of these functionalities, some complete tutorials, both written and video, would be very useful.

Apart from that I want to thank you for the work you do. The more I use the keyboard, the more I get used to it, and I think that when the software part (especially Chrisalys) will be more complete, it will be very difficult to recommend another keyboard.

gedankenexperimenter, 2021-06-02:
There is now a PR for a [new AutoShift plugin](https://github.com/keyboardio/Kaleidoscope/pull/1036) for Kaleidoscope.  I don't know if and when it might be merged or included in an official Chrysalis build, but it's available for you to try out if you build your own Kaleidoscope firmware.

robleach, 2022-10-19:
Any time frame for when this might make it into Chrysalis?

necsoiu, 2022-10-19:
> Any time frame for when this might make it into Chrysalis?

I have the same question. I think this is a very important and long expected feature.

robleach, 2022-10-19:
> Any time frame for when this might make it into Chrysalis?

If there is anyway that I can assist, or help doing the work, I'm willing. What can I do to help?

robleach, 2022-10-31:
Been playing around with the PR build bits, and things appear to be working fine.

robleach, 2023-09-23:
Any chance Autoshift is going get some love and merged in so it's supported in the release builds? I just updated my firmware and noticed it's still not supported yet. So that means I need to choose to either downgrade again, or lose that functionality.

obra, 2023-09-25:
@robleach Are you asking about the release builds of Kaleidoscope or of Chrysalis? (AutoShift should be working in Chrysalis if you have the support in your Kaleidoscope build)

I don't have an ETA for AutoShift enabled in the Kaleidoscope firmware builds by default - Right now, my software time is going into porting Chrysalis to run purely in browser - We've had too many issues with reliability around the electron version of the app and "flashing firmware updates just works" is a prerequisite for changes that are going to require EEPROM layout updates.

robleach, 2023-09-26:
Okay that was a good question. I assumed it was Chrysalis, but I was mistaken it appears. So thanks for clarifying. Then I guess this issue could be close if Chrysalis now supports Autoshift by default, and now we're just waiting on support from the firmware build. Well at least now I know I don't have to downgrade Chrysalis. Thanks!