keyboardio/Chrysalis #686: Download badges are missing

fvaneerden, 2021-04-21:
Small issue: under the [installation](https://github.com/keyboardio/Chrysalis#from-installer-packages) section the following is written:
```
From installer packages

Click on the badge for your operating system's packaging system above
```
The badges seemed to have disappeared though...

algernon, 2021-04-21:
They seem to be still available in the [Downloads](https://github.com/keyboardio/Chrysalis#downloads) section, just below the screenshot at the top. Perhaps we should link to that section in the installation part, so the browser scrolls up...

algernon, 2021-04-21:
After adding a link, I re-read the section, and it was, indeed outdated, as it referred to per-os badges, which we no longer provide, but link to the appropriate GitHub release instead. I adjusted the section accordingly.

Thank you!

closed 2021-04-21 by algernon

fvaneerden, 2021-04-21:
Thanks for the quick fix!

commit a1dcb3d7 (2021-04-21): README.md: Link to the downloads in the installation section

Fixes #686.