keyboardio/Chrysalis #694: Your computer won't let Chrysalis talk to your keyboard.

benschinn, 2021-05-25:
**Describe the bug**
When using my planck rev6 keyboard Chrysalis cannot read the keyboard. I have tried to fix this my granting access via security setting on system settings to no avail. I am not familiar enough with Chrysalis development to dig understand the error message. Please advise!

**To Reproduce**
Steps to reproduce the behavior:
1. Open Chrysalis
2. An Error message pops up saying my computer won't let Chrysalis talk to keyboard
3. I click the fix it button.
4. More error messages pops up
5. dev console is clear

**Expected behavior**
Be able to read keyboard.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: macOS
 - Chrysalis Version: Version 0.8.2 (0.8.2)

**Additional context**
I have used Chrysalis without issue with my Model01.

algernon, 2021-05-25:
There are two issues in play here: one is that the error message is erroneously displayed on macOS (it's a Linux specific thing), the second is that Planck rev6 is sadly not supported by Chrysalis yet. Kaleidoscope (the firmware powering keyboards supported by Chrysalis) only support older Plancks with an ATmega32U4/Teensy MCU, the newer revisions that use an ARM-based MCU aren't supported yet. That's why Chrysalis can't talk to the keyboard: it needs to be running Kaleidoscope first.

Sadly, Chrysalis can't reliably distinguish between supported and unsupported Plancks, I'm afraid.

benschinn, 2021-05-26:
Thank you for the explanation! Closing the issue.

closed 2021-05-26 by benschinn