keyboardio/Chrysalis #697: Space cadet not working

DeliciousRoastPotato, 2021-06-17:
**Describe the bug**
I am using Chrysalis with my Keyboardio Model 01. I have set Chrysalis up to use space cadet for the shift keys. Another key is set up to turn space cadet on and a fourth key to turn space cadet off. However, despite all this, space cadet is not inserting parentheses when I press shift. 

**To Reproduce**
The log is attached below along with screenshots. 

**Expected behavior**
I expect that if space cadet is turned on, then pressing shift without pressing any other key should yield a ( or ) for left and right shift respectively. That is not happening. 

**Screenshots**
If applicable, add screenshots to help explain your problem. If possible, please also include the debug bundle you can create on the "System Information" screen. If, for some reason, you are unable to access that screen, please press `Control + Shift + I` to bring up the developer console. Right-clicking inside it will allow you to save its contents to a file - please attach that to the bug report.

**Chrysalis bundle**
{
  "timestamp": "2021-06-17T10:46:31.115Z",
  "uuid": "76e64156-8c5a-4f72-9ae7-65ab8e87057f",
  "chrysalis": {"version": "0.8.4"},
  "os": {
    "platform": "win32",
    "distro": "Microsoft Windows 10 Home",
    "release": "10.0.19042",
    "codename": "",
    "kernel": "10.0.19042",
    "arch": "x64",
    "hostname": "DESKTOP-QUNEKIC",
    "fqdn": "DESKTOP-QUNEKIC",
    "codepage": " 437",
    "logofile": "windows",
    "serial": "00325-96086-34263-AAOEM",
    "build": "19042",
    "servicepack": "0.0",
    "uefi": false,
    "hypervisor": false,
    "remoteSession": false
  },
  "logs": [
    {
      "level": "debug",
      "message": [
        "focus.find: portList:",
        [
          {
            "path": "COM3",
            "manufacturer": "Intel",
            "pnpId": "PCI\\VEN_8086&DEV_9D3D&SUBSYS_06DC1028&REV_21\\3&11583659&0&B3",
            "locationId": "PCI bus 0, device 22, function 3"
          },
          {
            "path": "COM4",
            "manufacturer": "Microsoft",
            "serialNumber": "Ckbio01E",
            "pnpId": "USB\\VID_1209&PID_2301&MI_00\\7&BC4F96A&0&0000",
            "locationId": "0000.0014.0000.003.001.000.000.000.000",
            "vendorId": "1209",
            "productId": "2301"
          }
        ]
      ]
    },
    {
      "level": "debug",
      "message": [
        "focus.find: found_devices:",
        [
          {
            "path": "COM4",
            "manufacturer": "Microsoft",
            "serialNumber": "Ckbio01E",
            "pnpId": "USB\\VID_1209&PID_2301&MI_00\\7&BC4F96A&0&0000",
            "locationId": "0000.0014.0000.003.001.000.000.000.000",
[... 3201 lines cut ...]
          15,
          15,
          15,
          15,
          15,
          15,
          15,
          15,
          15,
          15,
          15,
          15,
          15,
          15,
          15,
          15,
          15,
          15,
          15,
          15
        ]
      ]
    }
  }
}

gedankenexperimenter, 2021-06-17:
I've copied the JSON from above to a [gist of the Chrysalis config](https://gist.github.com/gedankenexperimenter/e664ceba278e0495933342280f9c3976); @DeliciousRoastPotato, would you mind editing the description above to remove it so it won't require so much scrolling to see replies here?

gedankenexperimenter, 2021-06-17:
My first thought is that maybe the firmware that you've got on your Model01 isn't the "experimental" firmware that includes the SpaceCadet plugin.

Olical, 2021-06-17:
I have the experimental firmware and Chrysalis 0.8.4, but I'm facing the same issue, if there's any data I can get you to help you diagnose it I'll happy grab it for you. Thanks for providing an awesome tool!

gedankenexperimenter, 2021-06-17:
SpaceCadet is disabled by default in the experimental firmware. Are you using a `Key_SpaceCadetEnable` key in your keymap to turn it on?

Olical, 2021-06-17:
Ah I see, I thought I needed experimental but I've tried both and get the same results. And yep, bound the key and I've mashed it a fair bit, no change I'm afraid. Pressing the key while in `xev` doesn't display anything, so it's definitely bound to something other than the default macro (I put it on the `any` key).

gedankenexperimenter, 2021-06-17:
You do need the "experimental" firmware for SpaceCadet.  I've looked through the code, and I think I've found the problem: the SpaceCadet enable & disable keys are off by one from what they should be.  Try using the SpaceCadet _disable_ key, which I'm pretty sure is actually the enable key.

Olical, 2021-06-17:
Oh sorry, I misunderstood entirely, using experimental and the disable key now and it's working! Great job spotting that! Thank you so much for the help, hopefully this was the same issue for the original creator of the issue.

gedankenexperimenter, 2021-06-17:
Thanks for testing it.  I'll submit a PR to get it fixed properly.

PR #698 (2021-06-17): Fix SpaceCadet enable and disable key values
The keycode values for the SpaceCadet enable and disable keys were off by one, such that the "disable" key was actually the _enable_ key, and the "enable" key was in the Steno range.

Fixes #697
closed 2021-06-17 by algernon