keyboardio/Chrysalis #699: Flatpak

jwillikers, 2021-07-25:
**Is your feature request related to a problem? Please describe.**

AppImage is a great distribution format, but isn't the easiest to use in terms of automatic updates, desktop integration, and installation directly from software centers.
It would also be nice to have some sandboxing around the application if possible.

**Describe the solution you'd like**

I would love to be able to easily install Chrysalis on Linux as a Flatpak from [Flathub](https://flathub.org/home), allowing me to get desktop integration, automatic updates, and sandboxing while also being easier for me to install.

**Describe alternatives you've considered**

Snaps offer many of the same advantages and AppImages can be extended to add most of the desired features, but the latter requires a lot more configuration of the separate tooling.
Snapd, the service required to use snaps, is really only installed by default on Ubuntu though it is available on other distributions. The distributions I use, i.e. Fedora and elementaryOS, have Flatpak support baked-in out-of-the-box.
Though, Snaps have supported packaging electron applications for a while now.

**Additional context**

Flathub hosts repositories for the applications they distribute which contain the necessary files to package the application, which can be a single file, the Flatpak manifest.
Ideally, the application developers open a PR to Flathub's repository with the necessary files where they will be given write access when the PR is accepted and the repository is created.
Earlier this year, Flatpak added support for packaging Electron applications, making this viable for Chrysalis.
Electron Builder should generate most of the necessary files for the Flatpak, though I believe an AppStream metadata file will be necessary either in Chrysalis's repository or as part of the Chrysalis Flathub repository.
Additionally, some applications choose to maintain a development Flatpak manifest in their own repository for making local development builds and/or nightly builds using the latest source.

algernon, 2021-08-02:
I am completely unfamiliar with flatpak, so I'm not sure when I can get around to work on this, as it'd involve learning at least what AppStream metadata files are, and how to create one for Chrysalis. 

However, I'm open to including such a file in the Chrysalis repo, and tweaking our build and publish process to create a flatpak app too, perhaps even to push it to flathub automatically (on release, or perhaps even for dev builds, if flathub supports different channels).

It is unlikely I'll get around to this myself in the near future, so any help would be most appreciated.

jwillikers, 2021-08-02:
Awesome, I'm glad to hear you're up for this. I can definitely help with that. I've done some work with app developers to Flatpak their applications and get them on Flathub recently. I haven't done an Electron-based app yet, though, so it'll take me some time. I'm sure I can spend some time putting together that AppStream metadatafile, too.

As for your build / release process, if you're app is published on Flathub, their CI system will hand building your application. Once you're app is accepted on Flathub, you get push access to a repository under their organization on GitHub for your app. This repository has the Flatpak manifest used to build your application Flatpak and necessary dependencies. That manifest will be tied to a particular release of Chrysalis, so you'll just update the release (and anything else as needed) and it will automatically be built and published to Flathub.

Projects that include a Flatpak manifest with their source code typically have it build from the latest commit on master instead of an exact release, which allows for things like nightly builds and building the latest version locally. Some projects go so far as to develop locally on Linux using Flatpak to reduce the complexity of the various Linux distributions. So you can incorporate it into your app as much or as little as you like.

There are two more things to keep in mind. First, you will want to add release information to the AppStream  when ever you cut a new release. This info will be shown to the end user whenever they update or view the app on an AppStore. Second, you will probably want to validate the AppStream metadata file and `.desktop` files, at least before a release, so that you don't have a small typo blocking your update on Flathub which would require another release just to fix.

algernon, 2021-08-02:
>  I'm sure I can spend some time putting together that AppStream metadatafile, too.

\o/

> Once you're app is accepted on Flathub, you get push access to a repository under their organization on GitHub for your app. 

Nice. We can then just have an "update & publish to flathub" workflow in our CI, and do the necessary steps, then let flathub handle the rest.

> Projects that include a Flatpak manifest with their source code typically have it build from the latest commit on master instead of an exact release, which allows for things like nightly builds and building the latest version locally. 

We've been doing release and snapshot builds, and I'll want to keep that going forward. So for flathub, I'd want to ideally publish both, on separate channels. If that's not a practical thing to do, then release builds only. Snapshot builds aren't something I want our users to automatically follow, they should be opt-in. We want our users to run release-quality code by default, with the option to run development builds if they need a feature or a fix ASAP. But even then, going back to a release build would be the recommended way for end-users.

For developers, running locally via `yarn run start` usually provides a better experience, with auto-updating app as the source changes (without restarting) and similarly convenient stuff.

> There are two more things to keep in mind. First, you will want to add release information to the AppStream whenever you cut a new release. This info will be shown to the end user whenever they [...]

That should be no problem.

jwillikers, 2021-08-04:
That all sounds great to me, except that I think I caused some confusion around including a Flatpak manifest in your project and having one on Flathub. A lot of projects have *both* and I'll describe the difference as best I can here.

The one on Flathub will end up in a repository named after your app-id, i.e. `github.com/flathub/io.keyboard.Chrysalis`. The manifest here would be named `io.keyboard.Chrysalis.yaml` and always target releases. In fact, the particular release is hard-coded in this manifest. Flathub will build your Flatpak for from this manifest make it available to end users. Technically, this will be made available on the `stable` channel which Flathub uses as its primary channel. They also appear to have a `beta` channel that apps can also use, which would be perfect for snapshots in your current release-flow. Users would have to specifically add the Beta Flathub repository and then selectively install the app from the Beta channel, something only power-users are ever likely to do.

A Flatpak manifest stored in this repository, `github.com/keyboardio/Chrysalis`, won't factor in a wit to Flathub or the releases thereon. This manifest usually includes `.Devel` as part of the file name and app-id, i.e. `io.keyboard.Chrysalis.Devel.yaml`. Projects usually include such a manifest to make it easier for developers to build a Flatpak from the latest source code locally and/or to host their own *nightly* builds, like GNOME and KDE. These organizations host their own Flatpak repositories to make these nightly Flatpaks available, and they are not related to the ones on Flathub. Development builds like this will usually differentiate between the development version and the release version from Flathub so that developers can even have both versions installed simultaneously without confusion.

jwillikers, 2022-10-04:
Just to follow up, I can easily set up a `beta` branch on Flathub once the app is published, which can be used to track pre-releases. I'll probably just need to configure a little bit of automation to update the `generated-sources.json` from the `yarn.lock` file when an update is detected and perhaps add an entry to the AppStream metadata for the pre-release version.

I'll remove the Flatpak manifest in `build/io.keyboard.Chrysalis.Devel.yaml` and just keep the one as part of the Flathub repository to reduce the potential for configuration drift.

obra, 2024-02-26:
Chrysalis now runs in-browser using WebSerial and WebUSB, so I'm closing out this issue as obsolete.

closed 2024-02-26 by obra