keyboardio/Chrysalis #700: Chrysalis fails to run or provide a useful error message with a broken udev setup.

obra, 2021-07-30:
Reported by a customer:

Hello, Jesse,

I'm not sure if you remember our email discussion of my failed
attempts (back in June) to run Chrysalis inside an "Lxroot" virtual
environment to configure my Keyboardio Atreus.

I did some more testing today.  One of the cool things about Lxroot is
that Lxroot can build a chroot environment (i.e. a custom mount
namespace) by bind-mounting together a subset of the directories from
the host system.

After about 15 minutes of testing, starting with a complete set of
bind-mounted directories on a working host (a Live-USB boot of Ubuntu
20.04), and then removing (one at a time) various bind-mounted
directories from the Lxroot mount namespace, I believe I have
identified the problem.

The problem I encountered previously was due to /run/udev not being
present in the Lxroot environment in which Chrysalis was running.

When I bind mount /run/udev from the host mount namespace into the
Lrxoot mount namespace, then Chrysalis appears to run successfully, at
least thus far.

It also appears that /run/udev is the only part of /run that Chrysalis needs.

I will leave it to you to decide whether or not it is worth printing
any kind of error message when /run/udev is absent.  And/or whether or
not Chrysalis should be able to run without /run/udev.

parke, 2021-07-30:
Chrysalis runs when `/run/udev` is absent.  However, in this case, Chrysalis fails to do anything useful.

When `/run/udev` is absent, Chrysalis displays the following message:

    No keyboards found.  Please connect your keyboard, then click "Scan"

The above occurs even when the keyboard is connected and `/dev/ttyACM0` exists and has permissions `666`.

If you want assistance reproducing this problem, let me know.  I may be able to provide more information and/or a script that reproduces the problem.

algernon, 2021-07-30:
We can certainly check the presence of `/run/udev`, and display a more helpful message. Would something like this work:

> Chrysalis needs access to `/run/udev` to detect connected keyboards, but it appears, it does not. Please make sure Chrysalis sees, and can read said path.

The reason - I suspect - we need `/run/udev` is for scanning for devices. Once we scanned, we probably don't need it. Which just gave me another idea: what if Chrysalis offered a way to try and connect to a device the user specifies? So if no keyboards are found, and `/run/udev` is unavailable, we'd present a message:

> Chrysalis was not able to detect keyboards, because it does not have access to `/run/udev`, or it is not available. However, if you know the device path of the keyboard, you can type that below, and connect without detecting keyboards:
> [_______]

parke, 2021-07-30:
I'd suggest plainer error messages:

 > Error reading from `/run/udev/<path>`.  Due to this error, Chrysalis is unable to auto-detect your keyboard.

> No keyboards found.  Would you like to manually specify the device of your keyboard?

algernon, 2022-06-24:
Been a while I commented on this, but I'm doing another pass at finding low-hanging fruits, and this is one of those - if we handle the reporting part only.

I do not think we should offer a "direct connect" option at this time, so I'll just make Chrysalis report a more useful error message instead.

PR #962 (2022-06-25): linux: If /run/udev is not accessible, give a useful error message
We need `/run/udev` to be able to scan devices, so if it is not accessible, or doesn't exist, we will now display a useful error message.

Fixes #700.

[image]
commit b4b66a8e (2022-06-25): linux: If /run/udev is not accessible, give a useful error message

We need `/run/udev` to be able to scan devices, so if it is not accessible, or
doesn't exist, we will now display a useful error message.

Fixes #700.
closed 2022-06-25 by obra