keyboardio/Chrysalis #705: Once toggled, the Shift modifier on a Shift key cannot be untoggled.

parke, 2021-08-03:
In Chrysalis 0.8.4, select a Shift key.  Then toggle on the Shift modifier for that key.  They key will now be "S+ Shift" (which, admittedly, does not make a lot of sense).  However, it will now be impossible to untoggle the Shift modifier.

The only solution I have found is to reassign the key to something other than Shift, then remove the Shift modifier, then reassign the key back to Shift.

Perhaps it should not be possible to apply a Shift modifier to a Shift key?

It feels really strange to be able to turn the toggle on, but not off.

The same thing seems to happen with Control.  (Perhaps it will also happen with Alt, Super, and AltGr, but I have not tested those.)

gedankenexperimenter, 2021-08-03:
While the behaviour described is certainly a UI bug, it should be quite harmless; Kaleidoscope has no problem with modifier flags on modifier keys, even if the flag duplicates the keycode.  Does it only happen when applying the `shift` modifier to the _left_ `shift` key, or does it also happen with _right_ shift?

parke, 2021-08-03:
It appears the right Shift key is not affected.

TrueFalcon, 2021-08-04:
Another way to reset the key is to just click the X at top left in front of "LAYOUT EDITOR" and confirm that you are OK with losing your unsaved changes.

commit 6036161d (2022-05-17): Editor: Do not allow augmenting a modifier with itself

When configuring a modifier key, do not allow augmenting it with itself. We
already disallowed this when the base keycode was the modifier, but we need to
disallow the case where the current code is the modifier in question, too.

Fixes #705 by not allowing the situation to occur in the first place.
commit b15dfab8 (2022-05-17): Editor: Do not allow augmenting a modifier with itself

When configuring a modifier key, do not allow augmenting it with itself. We
already disallowed this when the base keycode was the modifier, but we need to
disallow the case where the current code is the modifier in question, too.

Fixes #705 by not allowing the situation to occur in the first place.
closed 2022-05-17 by algernon