keyboardio/Chrysalis #712: Use Layout Editor without having the keyboard connected

jimmysjolund, 2021-08-18:
I'm waiting to receive an Atreus and thought I would prepare my layout while waiting. However it seems I can not access the Layout Editor without having a keyboard connected? In my experience I will plan and mess around with my layout for quite some time before flashing. It would be beneficial if I could work on my layout in the editor on different computers (for instance) and once I'm ready, connect the keyboard and flash it. Now I have to wait until I get the keyboard and it has to be connected the entire time I change around layouts and try out different ideas.

obra, 2021-08-18:
This is indeed a desirable feature, but we don't currently have an
estimated schedule for it.

algernon, 2022-06-26:
This is the same as #377, so closing this in favour of the older request.

closed 2022-06-26 by algernon