keyboardio/Chrysalis #713: Chrysalis starts with a blank white screen - (undefined symbol / bindings.js)

Comradin, 2021-08-20:
**Describe the bug**
Upon startup Chrysalis displays a blank white screen. Opening the developer console I see two warnings and one error.
Looks like there is an undefined symbol in bindings.js according to the error message.

Maybe this gets introduced due to the ArchLinux AUR, which has a dependency for 'electron'. And electron just points to the latest release. So it could be, chrysalis isn't compatible with electron release 13?

**To Reproduce**
Steps to reproduce the behavior:
1. Start chrysalis
2. See blank screen

**Expected behavior**
The application worked before, 

**Screenshots**
```
internal/worker.js:82 [Deprecation] SharedArrayBuffer will require cross-origin isolation as of M92, around July 2021. See https://developer.chrome.com/blog/enabling-shared-array-buffer/ for more details.
(anonymous) @ internal/worker.js:82
/usr/lib/chrysalis/app.asar/node_modules/bindings/bindings.js:121 Uncaught Error: /tmp/.org.chromium.Chromium.QADEuL: undefined symbol: _ZN2v86Object18InternalFieldCountEv
    at process.func [as dlopen] (electron/js2c/asar_bundle.js:5)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1138)
    at Object.func [as .node] (electron/js2c/asar_bundle.js:5)
    at Module.load (internal/modules/cjs/loader.js:935)
    at Module._load (internal/modules/cjs/loader.js:776)
    at Function.f._load (electron/js2c/asar_bundle.js:5)
    at Function.o._load (electron/js2c/renderer_init.js:33)
    at Module.require (internal/modules/cjs/loader.js:959)
    at require (internal/modules/cjs/helpers.js:88)
    at bindings (/usr/lib/chrysalis/app.asar/node_modules/bindings/bindings.js:112)
electron/js2c/renderer_init.js:113 Electron Security Warning (Insecure Content-Security-Policy) This renderer process has either no Content Security
    Policy set or a policy with "unsafe-eval" enabled. This exposes users of
    this app to unnecessary security risks.

For more information and help, consult
https://electronjs.org/docs/tutorial/security.
This warning will not show up
once the app is packaged.
(anonymous) @ electron/js2c/renderer_init.js:113
```

**Desktop (please complete the following information):**
 - OS: ArchLinux
 - Chrysalis Version: 0.8.4-1 (aur/chrysalis)
 - Electron: 13.2.1-1

Comradin, 2021-08-20:
Just checked the aur/chrysalis-bin package, which has no electron dependency defined. Using this installation, the software starts just fine.

I have these three electron packages installed:
```
❯ yay -Qa | grep electron
electron 13.2.1-1
electron11 11.4.12-1
electron12 12.0.17-1
```

I guess, the aur/chrysalis-bin uses one of the other electron installations or is packaged with its own, which is not the v13?

obra, 2021-08-20:
I don’t think we maintain that aur. You may want to file a bug over on their side.

algernon, 2021-09-27:
As far as I understand, the AUR works, but the git checkout does not, making this our problem, kinda. The issue at hand is that Chrysalis uses some compiled node packages, an those need to be compiled for the same version of Electron as Chrysalis is running with. My educated guess is that these packages (serialport, in particular) is still compiled against Electron 12, while attempting to run Chrysals with Electron 13.

To resolve that, bumping the electron dependency in `package.json` to 13, and rebuilding should do the trick, hopefully.

We'll upgrade to electron 13 too, eventually.

algernon, 2022-05-20:
We have upgraded to Electron 17 since, but the problem will exist at least as long as we use compiled modules in the renderer process, possibly even after.

I don't think this is our bug, to be honest, it's just how Electron works. We are tied to a specific Electron version, and that is intentional, trying to use Chrysalis with another version is not supported, and if it doesn't work, that's not a bug.

Luckily, thanks to @obra's work, we can move to newer Electron versions considerably easier now, so we shouldn't lag behind _so_ much.

With that said, I'm closing this issue, as there's nothing more we can do.

closed 2022-05-20 by algernon