keyboardio/Chrysalis #727: Swedish key

alex-pradas, 2021-12-03:
**Describe the bug**
Swedish keyboard layout in

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem. If possible, please also include the debug bundle you can create on the "System Information" screen. If, for some reason, you are unable to access that screen, please press `Control + Shift + I` to bring up the developer console. Right-clicking inside it will allow you to save its contents to a file - please attach that to the bug report.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Chrysalis Version: [e.g. 0.4.0]

**Additional context**
Add any other context about the problem here.

algernon, 2021-12-03:
Can you try the snapshot build, please? This particular issue _should_ mostly be fixed there.

alex-pradas, 2021-12-05:
The latest snapshot (82) fixes the characters that are shifted. The characters using AltGr are still shown as `AGr+X` where X is the correct character of the base layer. It would be much nicer if the user culd see the indended character :)

algernon, 2021-12-05:
Mhm, that's sadly expected. Due to technical reasons, I can't easily fix AltGr: we're using the [CLDR](https://cldr.unicode.org/) layouts as-is, and those files are missing the altgr symbols, at least in the windows variants... We may be able to explicitly list which language uses which file from CLDR to have the best coverage, mind you, that's something we can try.

But if none of the [CLDR files](https://github.com/keyboardio/Chrysalis/tree/master/static/cldr) have the symbol, there's little we can do, except forking CLDR and/or contributing the symbols ourselves. That's not something we have the resources for, though.

algernon, 2022-06-29:
As it turns out: I was wrong. I was misinterpreting the CLDR data, it _does_ have the data, albeit we need to dig a little bit harder to get it out.

commit 845a2592 (2022-06-29): cldr: Pick up altR modifiers from a few more places

We can pick up altR-modified symbols from a few other keymaps, not just those
that have `altR` as the single modifier, but those that have `+caps?` too,
because caps is optional there.

This fixes #727.