keyboardio/Chrysalis #731: Apple Silicon/M1 build

Godatplay, 2021-12-30:
It'd be great to have a macOS build that supports Apple Silicon / M1 / aarch64.

obra, 2021-12-30:
The default macos build should support Apple Silicon machines. I agree that
it'd be ideal to have builds *optimized* for aarch64, but my recollection
is that that is blocked on github offering M1 based runners.

-j

algernon, 2021-12-30:
Theoretically, we can use custom runners. So if there's anyone who can offer us an M1 runner, maintain it, etc, we can make use of it, and don't have to wait for GitHub themselves to offer it.

Godatplay, 2021-12-30:
> The default macos build should support Apple Silicon machines.
Oh! By "default" do you mean the one on brew? The 0.8.3 Github build requires Rosetta

algernon, 2022-03-29:
While this was closed a while ago, with #764 merged, our builds should now be Universal apps, and as such, should support M1 too.

commit 0135f48c (2022-03-29): Switch macos to Universal builds.

We need to override @electron/universal with @spotxyz/universal
because https://github.com/electron/universal/issues/17 is
as-yet-unresolved.

Fixes #731
closed 2022-03-29 by algernon