keyboardio/Chrysalis #733: Atreus keyboard does't work in UEFI/BIOS

necsoiu, 2022-01-03:
**Describe the bug**
Atreus keyboard does't work in UEFI/BIOS.

 - Chrysalis Version: [e.g. 0.8.6]

obra, 2022-01-03:
Hi,

Can you confirm that you've updated the firmware of the keyboard using the
latest version of Chrysalis?

(You can find instructions about how to do that here:
https://shop.keyboard.io/pages/atreus-getting-started)

Best,
jesse

necsoiu, 2022-01-04:
> Hi, Can you confirm that you've updated the firmware of the keyboard using the latest version of Chrysalis? (You can find instructions about how to do that here: https://shop.keyboard.io/pages/atreus-getting-started) Best, jesse
> […](#)

Hi, 
Yes I can cofirm I have updated the firmware like in your recomandation.
[image]
The reconnect option was not working, but after unplug and reconect the keyboart everything seems ok, except the keyboard still does't work in UEFI/BIOS.

obra, 2022-01-04:
Can you tell me what device it's not working on?  (This was a known issue
with the firmware that the Atreus shipped with, but that has generally been
believed to be fixed and working fine in recent Kaleidoscope releases.

Thanks!

necsoiu, 2022-01-05:
> Can you tell me what device it's not working on? (This was a known issue with the firmware that the Atreus shipped with, but that has generally been believed to be fixed and working fine in recent Kaleidoscope releases. Thanks!
> […](#)

I

necsoiu, 2022-01-05:
It's not working on any device. I've tested on Lenovo and HP laptops, and now it is connected on HP Prodesk 600 G4 MT (with latest bios firmware).

necsoiu, 2022-01-13:
> > Can you tell me what device it's not working on? (This was a known issue with the firmware that the Atreus shipped with, but that has generally been believed to be fixed and working fine in recent Kaleidoscope releases. Thanks!
> > […](#)
> 
> I

Why is this threat closed? It was solved? If yes, please tell me what should I do becaus I still can't use my keyboard in UEFI/BIOS.

obra, 2022-01-13:
It looks like you closed it yourself. I'll happily reopen it.

necsoiu, 2022-01-14:
> I've been trying to reproduce this, and made some progress: the Atreus with the same firmware has worked fine on my PC (UEFI), and on my - pretty ancient - laptop (old BIOS). However, I found a neighbour with a laptop where the problem is reproducible.
> 
> We're trying to narrow it down further.

Today I will take the keyboard home and test it with another custom made PC (with an Asus motherboard).

necsoiu, 2022-01-16:
> I've been trying to reproduce this, and made some progress: the Atreus with the same firmware has worked fine on my PC (UEFI), and on my - pretty ancient - laptop (old BIOS). However, I found a neighbour with a laptop where the problem is reproducible.
> 
> We're trying to narrow it down further.

On my Asus UEFI it also works very well. It seems that the problem is only with some UEFI/BIOS.

studio1202, 2022-02-05:
Same problem here with on all of my 3 machines: on my Fujitsu laptops (Celsius H730 2013 and Lifebook E5410 2021) I cannot enter the disk password, while on a Dell Poweredge T20 server both keyboard and mouse work in the BIOS GUI but not in the boot selection menus. All firmwares are up-to-date.

gedankenexperimenter, 2022-02-07:
Quite a while back, while I was working on the HID module of Kaleidoglyph, I noticed a problem where Kaleidoscope wouldn't switch to 6KRO mode unless I made a particular change to KeyboardioHID.  It was not at all clear to me why that change worked, nor why the existing code didn't, so I posted a draft PR to try to get a discussion started: keyboardio/KeyboardioHID#50.  Alas, we never did pursue the cause then.  It might be related to this issue, but I haven't tested any version of it recently.

Snaptags, 2022-10-01:
Same for my Model 100 :-( Any news on the issue? That's kind of a dealbreaker for me, I don't want to keep my old keyboard on the desk just to login :-(

obra, 2024-02-26:
Closing out because this isn't a Chrysalis (configuration tool) issue

closed 2024-02-26 by obra