keyboardio/Chrysalis #735: Space key not repeating when keeping it pressed down

isipka, 2022-01-19:
Space key not repeating after pressing it and keeping it pressed down which is a problem as it is a frequent keybinding e.g.  jumping in games

**To Reproduce**
Steps to reproduce the behavior:
1. in any app press space key and keep it pressed
2. the space key is entered once and is not autorepeated

**Expected behavior**
When keeping the space key pressed the space character should autorepeat
If for some reason this is by design a user should be offered an option to change this behavior to autorepeat

**Screenshots**

**Desktop (please complete the following information):**
 - OS: Win 10
 - Chrysalis Version: 0.8.6

**Additional context**
Add any other context about the problem here.

obra, 2022-01-19:
This isn't the standard behavior for our keyboards or keyboard firmware, so
I'm pretty confused. Autorepeat is controlled by your PC, rather than the
keyboard.

Rom1deTroyes, 2022-01-19:
> 2\. the space key is entered once and is not autorepeated

Do you have a second action on this key ?
As I have a `layer shift when hold` on my <kbd>Space</kpd> key, it don"t auto-reapete, but other keys do...

I had addessed the problem by having a game layer, with all my needed keys and combo :-)

Happy hacking :wrench:

isipka, 2022-02-20:
> > 2. the space key is entered once and is not autorepeated
> 
> Do you have a second action on this key ? As I have a `layer shift when hold` on my Space key, it don"t auto-reapete, but other keys do...
> 
Yes I have

> I had addressed the problem by having a game layer, with all my needed keys and combo :-)
> 
Good idea, but it doesnt solve the problem of multiple presses on space with a layer shift as second action not working.
The autorepeat works though after pressing the space key twice in a row.

gedankenexperimenter, 2022-02-21:
This sounds like it is working as designed.  If you tap the `space` key, you'll get a single space (` `) character.  If you press and hold it, it will become a layer shift key once it times out (or until you tap another key).  If you tap it once, and immediately press and hold it, that sequence gets sent to the PC as a single press and hold of the `space` key, so you have a way to get auto-repeat of the ` ` character.

I'm not sure I understand what you mean by "the problem of multiple presses on space with a layer shift as second action not working", however.  Can you elaborate on that?

closed 2024-11-11 by obra