keyboardio/Chrysalis #737: Problem assigning tap/hold modifiers

cobblepot1, 2022-02-05:
**Describe the bug**
I cannot assign `Modifier when held` to multiple keys.

**To Reproduce**
Steps to reproduce the behavior:
1. I'm trying to set up home row mods for my Atreus, so for key A I added Secondary Action "Modifier when held" and set it to "Alt". No problem there. 
2. But trying the same thing with any other key/modifier combination deletes the layer #0 assigned main key and replaces it with #[object Object]. I've tried different combos of modifiers and only the first change to key A works. I also see  #[object Object] on the graphical keyboard display in Chrysalis

**Expected behavior**
I expect to be able to assign e.g. key `U` to type "u" when tapped and `Ctrl` when held down.

**Screenshots**
[image]
[chrysalis-debug.bundle.zip](https://github.com/keyboardio/Chrysalis/files/8007001/chrysalis-debug.bundle.zip)

**Desktop (please complete the following information):**
 - OS: Windows 10 x64
 - Chrysalis Version: 0.8.6

**Additional context**

obra, 2022-03-03:
> A user on Discord says the problem may be caused by an error in the
> Chrysalis keyboard mapping.
>
> https://discord.com/channels/492408953041321984/761681781447196693/945511438443753543
>
> Any progress on this is greatly appreciated!

@algernon tells me he'll be able to dig into this tonight. I'm sorry it's
been hanging out so long.

algernon, 2022-03-04:
Hi!

Apologies for this taking so long, I've been blessed with a whole lot of things that hindered my progress. The good news is, that I found the problem, and am working on a fix. It's a bit tricky, had a fix last night, but while that fixed this issue, it broke other cases. So I need to be a tiny bit smarter.

I'll have something ready soon!

quickfold, 2022-03-18:
Hi, it's 2 weeks since last update, and 6 weeks since the bug was submitted. I understand that the Model 100 is taking priority. When should I expect another update? Thanks.

lukeabbott, 2022-03-21:
Is this the reason my model 01 stops typing a key when I assign a secondary action?

gedankenexperimenter, 2022-03-21:
> Is this the reason my model 01 stops typing a key when I assign a secondary action?

Can you be more specific about the issue you're having?

lukeabbott, 2022-03-21:
In chrysalis, I attempted to assign a secondary action on hold to each of my home row keys. For example, on hold of 'a' register 'shift' modifier.

After assigning the secondary actions the keys stopped registering any key presses.

[image]

obra, 2022-03-21:
> Hi, it's 2 weeks since last update, and 6 weeks since the bug was submitted. I understand that the Model 100 is taking priority. When should I expect another update? Thanks.

I'm really sorry. @algernon has been out with a medical issue for most of that time. I don't have an ETA for when to expect a fix.

lukeabbott, 2022-03-21:
> I believe this is a different issue: the keys stop registering because the default firmware does not have the necessary plugin enabled.

Thanks!

quickfold, 2022-03-28:
@algernon 
> Are you using the dropdown to select the modifier, or the switches a block above? With the switches, I can reproduce something similar, though I do not get #[object Object] stuff, but random keycodes instead. That's the issue I was trying to fix up until now. But just when I was getting back to it today, I realized that the switches aren't meant to be used for this, the dropdown is. I can't repro with the dropdown:

I am using the dropdown. I think the issue is related to my use of the Dvorak layout. I see you have "English (US)" selected in the Linux layout above Secondary Action. Can you try with "Dvorak" selected? Mine is Windows layout with Dvorak selected. Thanks!

algernon, 2022-03-28:
Interesting. The layout should not make a difference, but I tried with "United States (Dvorak)" selected, and still couldn't reproduce with the dropdown. Still on Linux though, will see if I can reproduce on Windows (don't expect to, but it might surpruise me).

quickfold, 2022-04-04:
Hi @algernon , looking forward to your results from testing on Windows 10. I can recreate the bug on my machine and send you a video or debug information if you need.

quickfold, 2022-04-10:
Hi @algernon, I hope you don't mind the reminder. I know you are busy with other things. The original bug report was submitted Feb 5, over 2 months ago. I look forward to hearing more so I can use my Atreus. Thanks!

commit 40f81cf3 (2022-04-28): Editor/SideBar: Fix the code setting the secondary actions

`props.onKeyChange()` expects a keycode, not a key object, so pass it the code
of the new key. While looking up an object works most of the time, there are a
number of cases where it doesn't.

This should fix #737.
closed 2022-04-28 by algernon