keyboardio/Chrysalis #738: Sticky Shift Doesn't Work with Quote Key

jhitch, 2022-02-09:
I have a reproducible problem where my one-shot / sticky shift does not result in a shifted (double) quote. Holding the shift works as expected. One-shot works with all the alphanumeric keys. Just the quote key fails to get a one-shot / sticky shift no matter what I try.

MacOS 12.2
Build 0.8.3

gedankenexperimenter, 2022-02-11:
This sounds more like a Kaleidoscope bug than a Chrysalis bug.  I'll check it out sometime soon, possibly tomorrow.

gedankenexperimenter, 2022-02-11:
@jhitch — Please try the most recent firmware from the Chrysalis bundle (install Chrysalis 0.8.6, and use the Firmware Update tool). That version of the firmware has a completely new version of OneShot (the plugin that provides the "sticky" functionality).  If that works, please feel free to close this issue.

jhitch, 2022-02-11:
Got it. When having flashed the newest firmware, problem solved. Sorry should have tried that first.

closed 2022-02-11 by jhitch