keyboardio/Chrysalis #746: Use Shift and Secondary Action

pjaspers, 2022-03-01:
I want to map my left shift key so that it types a ( when pressed, and acts a shift key when held. I can't do this, because Secondary Actions are not selectable as soon as you use a shift (even when using the direct keycode).

Any ideas?

gedankenexperimenter, 2022-03-01:
You can do it, but you'll need to compile a custom Kaleidoscope sketch.  A `Key` value with modifier flags uses the same high bits as a "modifier when held" key uses, so you can't do it all in the keymap.  You can, however, define a Qukey object that will get you the behaviour you want.

cceckman, 2022-09-06:
+1 for this issue. This is a feature regression - "Space cadet" shift used to Just Work with stock sketches.

gedankenexperimenter, 2022-09-17:
SpaceCadet will also work, but also requires a custom sketch (and it always did; this is not a "regression").

algernon, 2022-09-17:
That's not entirely correct. SpaceCadet *is* part of the current default firmware sketches, for all keyboards, but it is disabled by default, and requires the "SpaceCadet enable" key to be hit to trigger it on - not very convenient, but the plugin is there. You are correct, however, that it is often more convenient to enable it by compiling custom firmware.

It *used* to be enabled by default, not just included, but we turned that off because it was a frequent cause of confusion.

Future firmware shipped with Chrysalis - for keyboards where we have space for it - will include the new `SpaceCadetConfig` plugin, which allows one to flip the plugin on on boot.

So, SpaceCadet is coming, which will help resolve this particular wish.

closed 2022-10-17 by obra