keyboardio/Chrysalis #752: LayerKeys should work like other sections

obra, 2022-03-23:
[image]

Instead of a dropdown and a text entry widget, we should group by type and enumerate all the layers, with one button per type per layer. We *may* want to fold the sections.

algernon, 2022-05-30:
While the original proposal has the *possibility* to reduce selecting a layer key to one click, as soon as we start folding sections, we'll be looking at at least two clicks, and a rather high "Layers & keymaps" section.

Instead of doing that, I'd propose keeping the type dropdown, but replacing the layer dropdown with enumerated keys instead. That way it always takes two clicks to set a layer key, but it is consistent, and we don't have to fold anything.

algernon, 2022-05-30:
(Continuing from [#873](https://github.com/keyboardio/Chrysalis/pull/873#issuecomment-1141527119)...)

> Big panels of numbered buttons feel _harder_ to understand to me. They use more space and are les obviously tied to the type selector above them.

If a big panel of numbered buttons is something harder to understand than the two dropdowns, I'm not sure I understood the original idea properly then, because that - to me - reads like even more buttons?

I do agree, mind you, that a big panel of numbered buttons isn't a good user experience, but I find it a tiny bit better - in the case of layers, at least - than two dropdowns. Mostly because we have a rather limited set of layers, more than 10 is going to be _rare_.

I'm not sure what would be a better solution. We don't want the two dropdowns, and we don't want potentially large set of numbered buttons. What else would fit?

I could imagine a dropdown + single button, where the button has an editable index, but I wouldn't want to use that, it feels horrible, even though it is compact.

I could maybe see [Grouped Avatars](https://mui.com/material-ui/react-avatar/#grouped) work, with a twist: Avatars would still be numbered buttons, but they'd collapse into a group when they reach a max group limit, and expand only on mouseover, maybe?

Or perhaps a [Speed Dial](https://mui.com/material-ui/react-speed-dial/)? Where the dial entries would be the layers?

Other than these, I don't really see how to do it nicely within the constraints of a sidebar. If we had right-click context menus on the keymap, that'd open up a few more options.

obra, 2022-05-30:
I think I was *wrong* when I opened the original issue.

> I could imagine a dropdown + single button, where the button has an
editable index, but I wouldn't want to use that, it feels horrible, even
though it is compact.

 That feels a lot like what we had before, just with different rendering on
the "button"

I wonder if the right answer is to break apart the *first* dropdown into
multiple different items.

algernon, 2022-05-30:
> I wonder if the right answer is to break apart the *first* dropdown into multiple different items.

So instead of a type dropdown, we'd have, say, type buttons? And keep the target layer dropdown?

obra, 2022-05-30:
> I wonder if the right answer is to break apart the *first* dropdown into
> multiple different items.
>
> So instead of a type dropdown, we'd have, say, type buttons? And keep the
> target layer dropdown?
I'm running a little bit slow today, but in my head, I think it *might*
work as a set of 5 big "buttons" that have text and a dropdown inside it.

algernon, 2022-05-31:
> I'm running a little bit slow today, but in my head, I think it *might* work as a set of 5 big "buttons" that have text and a dropdown inside it.

I can give that a try, but I don't think it's a good idea. Because depending on _where_ you click the button, different things will happen. If you click the dropdown, it drops down. If you click outside of the dropdown, it assigns the button. It feels similar to the type buttons + single dropdown idea, except we have more dropdowns, and they're inside clickable things.

Ooor... rather than 5 buttons, 5 sub-sections, each with a dropdown to select the target layer, and a single button to assign the combination of layer + type-subsection onto the key. Or a variation of it: one single button on the top, which serves as a thing to assign with. Below, 5 sub-sections, with a dropdown each. This way we have 5 dropdowns, 1 button, and we can  have some explanatory text inbetween.

obra, 2022-05-31:
That design doesn't sound right to me.

I'm happy to take a pass at this in the not too distant future.

algernon, 2022-05-31:
I'll leave it to you then, curious what you'll come up with!