keyboardio/Chrysalis #795: Chrysalis 0.9.0 wrote the wrong keyboard layout on Keyboardio Atreus during firmware update

rjc, 2022-05-18:
**Describe the bug**
Updating firmware on my _Keyboardio Atreus_ completely messed up my keyboard layout - the columns were shifted, i.e. _Q_ key produced _W_ when pressed, _C_ produced _V_, etc. It also obliterated all of my custom configuration. Is it possible that it detected and/or used the layout for _Model 01_?

**To Reproduce**
Steps to reproduce the behavior:
1. Plug in _Keyboardio Atreus_.
2. Open _Chrysalis_.
3. Setect a keyboard.
4. Connect.
5. Firmware Update.
6. Selected firmware - Chrysalis 0.9.0 default.
7. Press bottom left key.
8. Update.

**Expected behavior**
Firmware updates **without** touching the keyboard layout, including local modifications, unless _Restore to factory defaults when flashing_ explicitly selected.

**Desktop:**
 - OS: macOS 10.15.7
 - Chrysalis Version: 0.9.0

**Additional context**
Yes, I'm aware that version _0.9.1_ has been released shortly afterwards but that's what was, and (so far) still is, available via _homebrew_ - also, looking at the commits between the two versions, I see that the changes are cosmetic, i.e. _JS_ and _CSS_.

I have another Atreus which I can update the firmware on using this or another machine (macOS 12.4) with the same or newer version of _Chrysalis_ if need be.

BTW, I had to reset EEPROM to factory defaults and redo my custom layout by hand - I exported it to a file now (attached). Debug bundle also attached.

[rjc-atreus_custom_layout.json.txt](https://github.com/keyboardio/Chrysalis/files/8719372/rjc-atreus_custom_layout.json.txt)
[rjc-chrysalis-debug.bundle.json.txt](https://github.com/keyboardio/Chrysalis/files/8719373/rjc-chrysalis-debug.bundle.json.txt)

obra, 2022-05-18:
Yikes. That's absolutely no good. Is that debug bundle from the session where things went wrong? (I'm not immediately seeing the badness happen in there, but may have missed it)

obra, 2022-05-18:
We've reproduced the problem and are discussing the right fix here: https://canary.discord.com/channels/492408953041321984/761681781447196693/976580603510136852

I'm sorry to say that there's not a way to recover a board that's got updated firmware other than flashing old firmware back onto it.

rjc, 2022-05-18:
@obra Sorry, I had to do the factory EEPROM reset and carry on with work so the debug bundle has been taken after I finished work - on the same machine, with the same keyboard, and the same version of Chrysalis, though.

As to future upgrade, what's the best way forward? Download an older version of Chrysalis and flash an older version of firmware onto it? Or is it safe/OK to simly pull the older version of the firmware directly from the [git repository](https://github.com/keyboardio/Chrysalis/tree/master/static/Keyboardio/Atreus)? Needless to say, I'd rather not go through the whole ordeal again with this or any of the other keyboards - another Atreus and Model 01.

I don't use Discord. Do you have a dedicated IRC channel or Discord<->IRC bridge by any chance? Either way, I'd appreciate if you could keep me in the loop here.

obra, 2022-05-18:
(We no longer have an IRC channel.)

At this point, the right solution is to sit tight for a day or two while we implement a fix in chrysalis and release it as 0.9.2.

closed 2022-05-19 by obra

obra, 2022-10-11:
Yikes. I'm so sorry. Is that debugging bundle from the same session where
Chrysalis misbehaved?

rjc, 2022-10-11:
> Yikes. I'm so sorry. Is that debugging bundle from the same session where Chrysalis misbehaved?

Hi @obra

[You've already asked that question](https://github.com/keyboardio/Chrysalis/issues/795#issuecomment-1130481350) :^P
[No, it is not](https://github.com/keyboardio/Chrysalis/issues/795#issuecomment-1130607758)

No worries, we're all busy! ;^)

PR #796 (2022-05-19): Structured EEPROM save & restore for all keyboards
A major problem - as highlighted in #795 - with our flashing process is that during the firmware update, if the EEPROM layout changes, bad things will happen. In the past we tried our best to not introduce layout changes in the firmware we ship with Chrysalis, but one such change slipped through.

On one hand, we could back that out, and re-add the feature without a layout change, but a whole lot of people already flashed those firmware. We'd break the eeprom layout for some people no matter what we do.

So instead of that, we chose to fix the [... cut]