keyboardio/Chrysalis #799: Code cleanup: Get rid of the *Bootloader descriptors

algernon, 2022-05-19:
For some of our hardware plugins, we have stuff like `Model100` and `Model100Bootloader` descriptors. This is so that we can handle the case where Chrysalis is supposed to connect to a keyboard in bootloader mode. In that case, we go to the firmware upgrade screen, and during flashing, we don't do anything fancy like trying to save & restore eeprom.

This works fine, but we have a bit of duplication. For that reason, when I added the Keyboardio Atreus plugin, I did not make an `AtreusBootloader` descriptor, but baked the flashing differences into its custom `flash` method. This also works fine, except that `flashSteps` is shared between normal and bootloader mode, and that's confusing. Even if we do not end up doing all the steps there, it looks wrong and confusing on the screen.

I prefer having a single descriptor, so instead of splitting out an `AtreusBootloader`, I'd rather turn `flashSteps` into a function, so like `flash`, it can decide what steps to show, based on what mode the keyboard is in.

algernon, 2022-05-19:
This is proving to be a bit more difficult than anticipated. There are too many assumptions around the code base that are based on the Foo/FooBootloader split, and I'd need to remove them all before merging the bootloader descriptors.

In the short run, it probably makes more sense to split `AtreusBootloader` out instead.

obra, 2022-05-19:
Let's do that instead, then.

algernon, 2022-05-20:
I went to look into lifting out `AtreusBootloader`, and noticed that the only keyboard we use `*Bootloader` for is the Model100, and we only do so because the Model100 bootloader isn't on a serial port, so we need to handle it a bit differently. Further digging exposed 956e3fd5afcf314684e41cf0eea51adbeb66db84, the commit where I merged the `*Bootloader` descriptors into the main ones.

The commit message there makes sense. The Model100 has a workaround, and it's rather awkward: the Model100Bootloader descriptor's device PID and bootloader PID are set to be the same, while for all other keyboards, we simply have the device PID and bootloader PID in the same descriptor. Cleaner, nicer.

So I'd rather figure out how to fix the `flashSteps` issues, which is something we want to do _anyway_, see #811.

algernon, 2022-07-27:
I'm closing this one in favour of #1015, because that one's scope includes this, and more.

closed 2022-07-27 by algernon