keyboardio/Chrysalis #806: Clicking the "Backup & Restore" button on the Layout screen permanently freezes the UI

clemensw, 2022-05-20:
**Describe the bug**
Clicking the "Backup & Restore" button on the Layout screen permanently freezes the UI. (prior to flashing firmware with ensuing backup)

**To Reproduce**
Steps to reproduce the behavior:
1. Start Chrysalis v0.9.2. Chrysalis detects Model-01 and shows the Layout and Colormap Editor
2. Push the "Backup & Restore Button

Button changes color, but view does not change.
No more clicks on the App are registered.
Application can be quit via the menu.

This bug is not reproducable after the first time firmware was flashed in this version.  In that case the Backup & Restore view is displayed, offering to restore the backup made during firmware upgrade: Possibly the new view is not handling the case of no backups well?

**Expected behavior**
App switches to Backup & restore view.

**Desktop (please complete the following information):**
 - OS: MacOS 12.3.1
 - Chrysalis Version: 0.9.2

**Additional context**
Add any other context about the problem here.

algernon, 2022-05-20:
Indeed, it does not handle that case properly. We have a fix pending, it'll be in 0.9.3, and I'll push it to master later today, too, so it will be available in the snapshot builds.

algernon, 2022-05-20:
#807 contains a fix for this issue aswell, along with the rest of the things it does.

PR #807 (2022-05-20): backups: Make automatic EEPROM backups device-type specific
We do not want to offer backups made from a Model01 when on an Atreus. To this end, save backups in a vendor/product subdir, and load the library from that directory too.

While there, renamed the IPC functions to be more consistent, too.

Fixes #804, and also fixes #806.
closed 2022-05-20 by algernon