keyboardio/Chrysalis #810: The floating keyboard can sometimes cover other widgets, or move out of screen

algernon, 2022-05-20:
**Describe the bug**

The floating keyboard on the Editor screen is very, very nice, and works great when the window is sufficiently large, and we're connected to a smaller keyboard like the Atreus.

If the screen is smaller, or the keyboard is bigger, multiple problems can occur, which I'll present with screenshots:

### Model01 @ full screen

[image]

Even at a very reasonable window size, the Model01's palm keys are partially covered. It doesn't render anything unusable, but it doesn't look good.

### Model01 @ smaller window

It gets worse when we decrease the window size:

[image]

The floating keyboard is partially off-screen, and covers the sidebar too.

### Atreus @ smaller window

[image]

The problem persists with the Atreus too, at smaller window sizes

## Suggestions

We should probably set a minimum window size, prevent the floating keyboard from moving off-screen (if possible, I'd prevent it from moving into the sidebar too), and we should likely scale it down too if the window gets too small.

Rom1deTroyes, 2022-05-21:
Sorry if my proposition is out of scope (I don't know if the floating keyboard is designed to be always present on the editor screen), but maybe a close control button on the floating keyboard can be useful for one trapped in a situation where only the float keyboard is visible ?

The UI job is great, and wish I had time to help testing it !

obra, 2022-05-21:
(It's currently designed to always be visible.)

Rom1deTroyes, 2022-05-29:
I hijack once again this issue, this time as a reminder from discord discussion :

When the app window is resized, panes are resized, the Atreus keyboard is resized, but the picker stays where it was (if you cut the width by half, the numpad disappears).

Not a big deal when you realize that you can move it by click and dragging, but surprising enough ?

The UI enhancements are really great !

obra, 2024-11-11:
This UI is completely different in current chrysalis. resolving

closed 2024-11-11 by obra