keyboardio/Chrysalis #813: Crash on startup using the appImage

undu, 2022-05-21:
When I try to execute the AppImages for version 0.9.2 and 0.9.3 the window fails to appear. When I execute the binaries from the CLI the process crashes with this loglines:

```
$ ./Chrysalis-0.9.3.AppImage
[18747:0521/203410.302862:ERROR:gpu_init.cc(454)] Passthrough is not supported, GL is egl, ANGLE is
[18703:0521/203410.309855:ERROR:cursor_loader.cc(116)] Failed to load a platform cursor of type kNull
[18747:0100/000000.340865:ERROR:sandbox_linux.cc(377)] InitializeSandbox() called with multiple threads in process gpu-process.
fish: Job 1, './Chrysalis-0.9.3.AppImage' terminated by signal SIGSEGV (Address boundary error)
```

This happens when using Archlinux. As a sanity check I tried the 0.8.6 binary and it doesn't contain this issue.

algernon, 2022-05-21:
Huh. I think it's the first error that trips us over, and that might be an Electron thing. We upgraded from 12 to 17 between 0.8.6 and 0.9.x. A quick search suggests that we now require hardware acceleration.

Can you check if you have HW acceleration on? [This page](https://wiki.archlinux.org/title/Hardware_video_acceleration) seems to cover a lot of related things.

If it is off, and you can't enable it, I _may_ have a workaround. If you have it on, we'll have to dig deeper.

undu, 2022-05-21:
VA-API is working, not sure if the driver exposes enough to make electron happy: 
[vainfo.txt](https://github.com/keyboardio/Chrysalis/files/8748279/vainfo.txt)

undu, 2022-05-21:
I see on the wiki that chromium doesn't support hardware acceleration under wayland

undu, 2022-05-21:
As a workaround, setting the GDK_BACKEND environment variable to xorg works: `GDK_BACKEND=xorg ./Chrysalis-0.9.3.AppImage`

obra, 2022-05-21:
does https://wiki.archlinux.org/title/wayland#Per_application work for this?

undu, 2022-05-21:
Those parameters don't seem to have any effect, it still crashes.

commit de95df45 (2022-05-26): build/launcher.sh: Add a workaround for starting under Wayland

Unfortunately, Chromium's Wayland support is not working properly yet, and
Chrysalis fails to start under it. Force an xorg backend to avoid that.

The root cause seems to be that recent Electron requires GPU acceleration, and
Chromium under Wayland does not support that.

Fixes #813.
closed 2022-05-26 by algernon