keyboardio/Chrysalis #817: Extra stuff in backups causing some minor trouble on the Model 100

obra, 2022-05-24:
Restoring eeprom after flash should not send commands that:

1) don't appear in the device's help 

OR

2) don't appear in the backup

Backing up eeprom before flash shouldn't call commands that don't appear in the device's help

algernon, 2022-05-24:
With #805 merged, this shouldn't be the case anymore. Unknown commands would turn into noops:

[image]

See all the `(noop)`s. Those are all commands that aren't sent.

On the other hand, we currently do try to restore commands we don't have a backup for. We could guard against that, indeed.

However, our current problem is likely isn't caused by either of these. It seems to get stuck on `macros.map`, and I can reproduce the same thing on the Atreus too. Chances are the root cause is the same, so I'll look into why the Atreus fails, and hopefully that'll help the Model100 too.

algernon, 2022-05-24:
We _do_ send unknown commands during restore, however, because never call `focus.supported_commands()` after reconnecting. That's an easy fix, thankfully.

closed 2022-05-24 by obra