keyboardio/Chrysalis #843: Ideas to improve the Preferences screen

algernon, 2022-05-27:
While the Preferences screen has seen some great improvements recently (such as being turned into a tabbed page, thanks to @obra), it still feels rather dry, and not very user friendly.

We do not have adequate amounts of help text and descriptions to explain what each thing does. For example, a `[x] Dark Mode` toggle button just hanging there is neither descriptive, nor very elegant. The language selection dropdown with its darker background, and spanning the whole width of the panel isn't nice, either.

The "Developer tools" page is also very short on information. It's just two pretty non-descript toggles.

I propose we overhaul the whole thing, adding more help text. Rather than a `[x] Dark Mode` toggle, for example, we'd have a paragraph or two, and an `Enable / Disable` toggle, where the text changes depending on state, too. That feels way friendlier.

I have a couple of other ideas, and I'll be posting screenshots of mockups, once I get there, so we can discuss how to improve Preferences further.

obra, 2022-05-27:
I think we're probably looking for a sentence rather than a paragraph.  And
having the primary text change based on the toggle is...not necessarily a
good thing. It can get super confusing to manage.

I agree 100% that the ultra-wide dropdowns from the MUI guidelines don't
feel right. But the narrow dropdowns we have also feel bad.  I wonder if
there are desktop MUI apps that do this well. Or there are other MUI themes
that do.

Rom1deTroyes, 2022-05-27:
For the « Dark/Light theme » option, one of the best I've seen is from [postman](https://www.postman.com/) app :
[image]

The preview is explicit enough for not having a paragraph, and the sentence can be kept simple, because the pictures decribes well what it's all about :art:

commit bc530182 (2022-05-30): Major redesign of the Preferences screen

The Preferences screen has seen a couple of updates recently, but it was still
feeling very rough. This patch is an attempt to address that.

Every tab has been redone from the ground up, adding a lot of help text and
explanations, and dividing the various sub-sections by topic. While there, the
underlying code was often cleaned up, simplified, and modernized, common
components lifted out.

We no longer use `@api/focus`, but `activeDevice` from the global context, among
other things.

Fixes #843.
commit 011aaa4c (2022-05-30): Major redesign of the Preferences screen

The Preferences screen has seen a couple of updates recently, but it was still
feeling very rough. This patch is an attempt to address that.

Every tab has been redone from the ground up, adding a lot of help text and
explanations, and dividing the various sub-sections by topic. While there, the
underlying code was often cleaned up, simplified, and modernized, common
components lifted out.

We no longer use `@api/focus`, but `activeDevice` from the global context, among
other things.

Fixes #843.
PR #864 (2022-05-30): Complete redesign of the Preferences screen
This is a complete overhaul of the Preferences screen. The changes are numerous, so instead of explaining it all, I'll illustrate instead:

## User Interface

[image]

## My Keyboard

[image]

### Factory reset

[image]

## Developer Tools

[image]

# Summary & Notes

Every tab has been rearranged to place logical groups together, into their own sub-sections, neatly separated, with plentiful labels and hints.

Under the hood, the code has seen a lot of cleanup too: all direct uses of `@api/focus` is gone, we're using `activeDevice` from the global co [... cut]
closed 2022-05-30 by obra