keyboardio/Chrysalis #848: Focus API objects could use a bit of cleanup

obra, 2022-05-28:
We currently have global singletons of focus, db, and macros.  They don't feel *quite* right, but we don't yet know the right refactoring we want to do.

Discussion here: https://canary.discord.com/channels/492408953041321984/492409398493446146/979906046665584660

algernon, 2022-06-03:
Based on my recent experience with reworking `@api/log`, I have a proposal. But first, an explanation why we have the singletons in the first place.

In all three cases, we don't want multiple copies, but a single shared instance, because we in case of Focus and Macros, we also do modifications. We do the same with the KeymapDB too (when switching host layouts). So they all need to remain singletons. However, they don't need to be global, none of them.

We could just have `focus()` returning a focus instance, `focus().db()` a database, and `focus().macros()` the macros instance. All of these instances would be out of the global scope. This would also make our imports cleaner, because we would no longer need to import `@api/focus/keymap/db` or `@api/focus/macros`, but simply `@api/focus`, and ask focus itself for the stuff we need.

algernon, 2022-06-03:
We don't even need `focus()`, just `focus` is going to be enough.

Going through what all these packages do, I think the following public API would be our goal:

- `focus.db`: `.lookup()`, `.format()`, `.isInCategory()`, `.selectCategory()`, `.constants`
- `focus.layout`: (replacing parts of `KeymapDB`): `.getStandardLayout()`, `.setLayout()`, `.getLayoutLanguage()`, `.getSupportedLayouts()`, `.guiLabel`
- `focus.key`: (replacing parts of `Keymap` and `KeymapDB`): `.addModifier()`, `.removeModifier()`, `.addDualUseLayer`, `.addDualUseModifier`
- `focus.macros`: `.map`, `.storageSize`, and possibly some helpers to set/update/add individual macros and macro steps
- `focus.device`: `.findDevices()` (previously `focus.find()`), `.open()`, `.isAccessible()`, `.isSupported()`
- `focus.connection`: `.isOpen()`, `.close()`, `.getDescriptor()`, `.command()`, `.supported_commands()`, `.supported_plugins()`, `.getPath()`
- `focus.migrations`: `.hasLegacyCodes()`, `.migrateLegacyCodes()`

This would also address a big part of #890, I believe, and as far as I see, every use of the large number of methods focus & friends currently export.

Getting there won't be easy, as the entirety of `api/focus` needs to be redone from the ground up. But we'd get a cleaner api, more modern code, and something easier to maintain.

What do you think about the proposed APIs @obra?

obra, 2022-06-06:
Focus is the wire protocol. I think that the thing called "focus" should
really be the wire protocol that talks to the keyboard. Ideally, it should
be relatively dumb.

It makes sense to me to have things broken apart into:

* keyboard discovery
* making and maintaining connection to a single keyboard
* talking the focus wire protocol to a keyboard

* the host's keymap / layout and keycode database
* the database of keycodes our keyboards understand.
* working with a keyboard layout/keymap
* working with macros
* it does make sense to have "migrations" hang off the side.

If we're going to do this, it feels like the right time to make sure the
ActiveDevice abstraction is the only thing that react code talks to, rather
than trying to have react code send and listen for the results of focus
commands.

algernon, 2022-06-06:
Mhm. This does imply a very large overhaul of pretty much all of `src/api/`. While there, should we split it apart into parts that run in the main process, and parts that run in the renderer? Having `src/api` sit sometimes between the two, sometimes at both tables feels a bit weird and inappropriate, so clearer separation would be nice, and if we're rebuilding the whole lot, might aswell do the splitting at the same time to have a cleaner architecture.

The first three would go into the main process, the rest into the renderer, and the renderer would simply talk through ActiveDevice, which would do the necessary IPC calls.

Possibly out of scope, but if we had something we could stick into GlobalContext that'd hold the keyboard state for us, and be available globally, for any screen, that'd be huge, too. Perhaps we should make a KeyboardContext instead? It'd reduce traffic a _lot_, and make Chrysalis considerably smoother.

algernon, 2022-07-28:
While hammering away on #1015, I managed to dig myself into a deep `@api` hole, and spent many hours pondering about the whole thing, and how it should be reorganized.

Before we reorganize, however, we need to know what's wrong with the status quo. There are many problems. Before I get into details, an example: as part of my work in #1015, as of this writing the wip branch removes over [a hundred lines](https://github.com/keyboardio/Chrysalis/compare/master...focus/device-detection), while providing the same - or even more - functionality, and I only touched a small part of `@api`.

Now, lets dive into the details!

https://github.com/keyboardio/Chrysalis/blob/ff2ee9cfe03d83853dfd4037f1edbb8f4ba76769/src/api/hardware/index.js#L27-L30

The `@api/hardware` plugin exports two list, a couple of serial devices, and some non-serial ones. This makes little sense, and it doesn't really reflect what the difference is, and why some descriptors appear in both. This list is used by:

https://github.com/keyboardio/Chrysalis/blob/ff2ee9cfe03d83853dfd4037f1edbb8f4ba76769/src/renderer/utils/findKeyboards.js#L61-L88

And that's only a tiny part of it, there's `findNonSerialKeyboards()` above, and `focus.find()` too, both of which are a big, complicated mess.

The goal of all of these is to find supported keyboards. It should be a process that roughly goes: iterate over all USB devices, filter them down to ones with vid/pid pairs we recognise, store enough information about those that allows us to connect later. While going through that list, have a list at serial ports (our main communication channel with the keyboards), and adjust the USB device infos with data from the serial port list (ie, the path, mostly). The latter may not even be necessary.

Then, if we look at `@api/focus`, we see something that is much more than a wire protocol implmenetation. It originally had a plugin-ish architecture, allowed external packages to register new "commands". That's also something that is unnecessary at this point. It is also very closely tied to `serialport`, and does not support any other transport. It also has a *huge* API, with all kinds of functionality thrown in that doesn't really belong to a wire protocol implementation.

`focus.open()`, `focus.find()`, `focus.reconnectToKeyboard()`, `focus.addMethod()`, `focus.command()`, `focus.addCommand()` should all be somewhere else - if they should exist at all. Instead of shoving a lot of functionality into `@api/focus`, it should be a thin wire protocol implementation, and should be building more complex stuff - such as the command stuff - on top of it, in a different package.

The lack of clarity in `@api/focus`, the lack of conscious, thoughtful design has far reaching consequences. Because of the way `@api/focus` works, the way it detects devices, we have user-visible issues, such as not handling it well when two of the same kind of keyboard are connected: if you have two Model100s plugged in, and you try to flash one, Chrysalis will happily back up eeprom from the one you connected to, clear it's eeprom, reboot it and flash an update firmware - and then connect to the *other* one and restore the first one's EEPROM over it. Ooopsie.

This happens because `@api/focus` is unable to uniquely identify a keyboard. All it would take, really, is to remember the usb bus & device along with the vid/pid, and we'd have considerably fewer issues.

Another problem with the current `@api/focus` approach is that we can't easily differentiate between the same device in different states: we should be able to handle keyboards in their normal mode, handle them differently when they're in bootloader mode, and handle the case when they are in neither, and possibly run firmware that Chrysalis does not support. With the hodge-podge mess that is `focus.find()` and `findKeyboards()`, we can't do this.

Talking about far reaching consequences: precisely because `@api/focus` doesn't do a good job at providing tools to handle different states of a keyboard, we have a number of places that check `focus._port.path` to see if we're connected via serial or not. That's not only awkward, but wrong, too.

We also need to talk about how much of the things under `@api/focus` are singletons, and have a global object too. I don't think any of them should be singletons, really. A connected device should be its own object, rather than be a member of the `focus` singleton. We should pass the device object around, rather than relying on a focus singleton to handle that for us - doing so leads to stuff like `ActiveDevice`, which tries to make the situation less severe, but falls short when it comes to devices where we aren't *technically* connected in the sense Chrysalis treats connections: bootloaders for example. We only connect when we flash, so `ActiveDevice` does nothing for them, and `focus` doesn't know about them, either.

`ActiveDevice` is a good idea, mind you. It could serve as *the* source of truth, if it could handle all situations, all kinds of "connections", even if it isn't an active serial connection.

Then, we have all the stuff below `@api/focus`: the keymap db, macros, colormap, and so on. They really don't belong there, and the only reason they're there is because we never got rid of this plugin-like architecture in `@api/focus`.

[... cut ...] ltGr+S` in an *overlay only*. The lack of references, and the keymap database being a singleton feels very limiting.

We also suffer from the problem that both `@api/focus` and `@api/flash` are closely tied to a node environment, which makes it harder to move towards having a more secure render environment. It also makes it harder to support different kinds of transports in the future.

That's about the problems. In a later update, I will post my current ideas about how to improve the situation.

algernon, 2022-07-28:
So, onwards to some proposals!

I'm not going to go into details just yet, because I don't have everything figured out, and before I even start to think about the details, the goals and whatnot should be discussed first anyway. So what follows is a very high level look at how I think `@api` should look like.

At the core, I'd have the Hardware descriptors, which could be very similar to what we have now. Ideally, they'd be almost pure data, with the keymap SVG being the probable exception. Not much to change there.

Then, I'd like to have a transport layer - lacking a better name. This would be the thing that can give us a list of devices, can connect to one and give us back an opaque object corresponding to the connection, transfer data back and forth, handle disconnecting, and so on. Very low level stuff. This is `Serialport.new()` and a couple of things wrapped up in our own abstration. Supported devices should be uniquely identifiable: by USB bus + device and vid/pid pairs, for example, assuming an USB transport layer. This would allow us to support multiple keyboards of the same kind connected, without accidentally mixing them up during reboots.

On top of the transport layer, we'd have the protocol. Again, a rather thin wrapper that takes the data from the transport layer, and turns it into a very simple intermediate format. It should also serialize the same simple format to send back to the keyboard via the transport layer. Essentially, the focus protocol layer would just be an implementation of a request-response protocol. It would split up the data received from the keyboard, associate it with the request it's a response for, and that's it. It would not interpret the data further. It'd just provide a way to send a request, and receive a response, with the ability to bind the two together.

On top of the protocol, we'd have the features, a protocol-specific parser & serializer that takes the data from the protocol, and interprets it. This would be the thing that splits up the keymap into layers, that turns the raw data into structures Chrysalis can work with, and then can serialize it back. It should be reasonably high level, so thing that we usually handle together, would be the same "feature": for example keymap & colormap would be the same "feature", because they're both layer based, both are addressed similarly, and we usually handle them together anyway. The format the data is parsed into should be protocol-agnostic, however. While the parsing & serializing *is* going to be protocol specific, the format they work with, should be agnostic.

On top of features, we'd have the Keyboard layer. This would discover the features on the keyboard, and would serve as the API Chrysalis' GUI works with. It would query the data *once*, everything available, rather than doing ad-hoc queries - often multiple times - for the same thing. It would present the parsed configuration obtained from the keyboard. We could even keep a log of changes, so we could implement infinite undo. Though, I suspect that for the vast majority of cases, a simple "discard changes" method is fine. This layer could do that without talking to the keyboard: just save an original copy! Chrysalis would work with a clone, and if need be, could discard its copy, and ask this layer to give it a new copy. Saving back would replace the stored original with the modified one.

There are a few things not covered here, like how our keycode database fits into this whole architecture. That's intentional, and the answer is simple: it doesn't. The keycode database would be independent, that's an UI thing first and foremost, rather than something to shove into `@api`.

We should get rid of `@api` too, by the way. Apart from the transport layer, everything else can be purely renderer-side. If we use WebUSB for transport, then even the transport can be renderer-side. But until we get there, the transport can have one half in the main process, the other in the renderer, and voila, we have clear separation, we no longer randomly cross the node / renderer border, and can tackle #914, and move off of electron-builder, to a more modern build system.

This would require us to implement DFU ourselves, probably over WebUSB/WebDFU - thankfully, there are JS libraries for that! `dfu` and `node-usb` are both packages we could use, depending on our needs. As an interim solution, we could also move flashing to the main process, but that doesn't really fit into the architecture described above. It could be squeezed in, temporarily, if absolutely necessary.

algernon, 2022-07-30:
Another problem of the status quo: ActiveDevice is pretty much tied to a focus connection, it doesn't really exist for keyboards in bootloader mode, nor for non-Kaleidoscope keyboards, even if Chrysalis otherwise considers them connected.

obra, 2022-07-30:
(It should be the other way round. The focus connection should be tied to
an ActiveDevice.)

algernon, 2022-07-30:
It should be, yes. But the status quo is that if there's no focus connection, then ActiveDevice doesn't do anything useful. Ideally, ActiveDevice would hold information about the keyboard connected, whether via focus, or just virtually (in case of a bootloader, for example). Part of that information should be the connection object itself.

That's not the case yet.

obra, 2022-10-11:
> Mhm. This does imply a very large overhaul of pretty much all of src/api/.
> While there, should we split it apart into parts that run in the main
> process, and parts that run in the renderer? Having [...]

With the latest updates to USB, we *may* be able to shove all the discovery
back in the renderer, using the WebSerial and WebUSB API emulation in those
libraries. in a way that is  considered "OK" by Electron. That would get us
closer to running in a browser.

> The first three would go into the main process, the rest into the
> renderer, and the renderer would simply talk through ActiveDevice, which
> would do the necessary IPC calls.
>
> Possibly out of scope, [...]

...That's what ActiveDevice is supposed to be.