keyboardio/Chrysalis #849: 0.9.5 on Windows punts us back to the Keyboard selection screen

algernon, 2022-05-28:
We've received multiple reports that on at least Windows 10, connecting to a Model 100 does not work. The loading screen appears, but we're getting punted back to the keyboard selection screen without any error or warning. 0.9.4 did not have this problem, so it's a regression since.

The debug bundles provided to us aren't very helpful: they show that Chrysalis connects to the keyboard, there's communication back and forth, everything seems to be in order. Then, after a couple of commands, the next one we get is the debug log from the Keyboard Select screen's periodic scan.

I suspect this may be related to #830, because we had another issue, where we _did_ get an error message, about `activeDevice` not having a `.devicePath()` method, due to `activeDevice` being null. I fixed that in af6cee129b941c75bb726a7ca9840e1e33801522, but the problem still persists.

necdetaliozdur, 2022-05-28:
> We've received multiple reports that on at least Windows 10, connecting to a Model 100 does not work.

I just want to add that connecting to an Atreus does not work either. I'm on Windows 11. I reverted back to 0.9.4 which solved the issue.

algernon, 2022-05-28:
Reproduced on my Win10 VM: 0.9.4 works, can connect, 0.9.5 and later punts me back to the selection screen immediately. My windows build environment is still updating, it will take a while, but I'll go and bisect Chrysalis as soon as its done.

algernon, 2022-05-28:
Interestingly, bisecting ended up declaring aef6231c94f24cb76bd65621d6d6f41ede86aa9e as the culprit. That does not make much sense yet.

algernon, 2022-05-28:
Progress! Checking aef6231c94f24cb76bd65621d6d6f41ede86aa9e out, and **not** updating dependencies: we can connect. As soon as I run yarn, we break. However, checking out `package.json` and `yarn.lock` from the prior commit: we still break.

This suggests that my bisecting wasn't adequate. It also suggests that what we're looking for is yarn.lock changes between 0.9.4 and 0.9.5. The problem will be somewhere there.

algernon, 2022-05-28:
Yep. It's going to be one of the dependencies. `git checkout v0.9.4 -- package.json yarn.lock && yarn && yarn run start` on current master lets me connect.

algernon, 2022-05-28:
Looks like we ended up upgrading `usb` from 2.3.1 to 2.4.1. It appears the bug was introduced in usb 2.4.0, and fixed in 2.4.2 released ~21 hours ago. Upgrading to 2.4.2 fixes the problem on my VM.

commit 1b96e592 (2022-05-28): Upgrade `usb` from 2.3.1 to 2.4.2.

Previously, we were using `usb` 2.3.1, but in a recent spree of regenerating
`yarn.lock`, we ended up resolving the version to be installed to 2.4.1.
Unfortunately, `usb` 2.4.0 made the package context aware (a good thing), but
also broke it on Windows completely (not such a good thing).

Luckily, `usb` 2.4.2 was released some 21 hours ago, which fixes the regression
on Windows, and using it, Chrysalis can connect to keyboards on Windows again.

Fixes #849.
closed 2022-05-28 by algernon