keyboardio/Chrysalis #861: Ctrl-P force closes Chrysalis in Win10

rhomboid-shovel, 2022-05-29:
**Describe the bug**
Typing Ctrl-P in Chrysalis in Win10 causes a force-close

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to keyboard
2. Type Ctrl-P (from either Layout & Colormap Editor or Layout Card)
3. This results in Chrysalis terminating

**Expected behavior**
A print dialog, or nothing.

**Debug Bundle**
[chrysalis-debug.bundle.json.zip](https://github.com/keyboardio/Chrysalis/files/8793846/chrysalis-debug.bundle.json.zip)

**Desktop (please complete the following information):**
 - OS: Win10 Home 10H2 (19044.1706)
 - Chrysalis Version: 0.9.6-snapshot.221

algernon, 2022-05-30:
I've reproduced the problem.

Found an issue in the Electron bugtracker about printing crashing on linux with no printer present, so figured that might be the case on Windows too, but no: attaching a printer still crashes the process. So far, wading through the Electron bugtracker didn't yield any useful results.

I did manage to get a crash dump, so will try to see if I can discern anything from that.

algernon, 2022-05-30:
Interestingly, calling `window.print()` from the developer console doesn't crash.

algernon, 2022-05-30:
To narrow the problem down, I created a [small test case](https://github.com/algernon/electron-print-crash), to see if that works on Windows, because I don't see anything relevant in the Electron bug tracker, and I can't believe we're the only ones seeing the issue.

The test case crashes on Linux, too.

rhomboid-shovel, 2022-06-03:
fixed for me with snapshot.266

rhomboid-shovel, 2022-06-03:
Wow, I didn't know I had the power to close an issue. Hope I didn't screw anything up!

closed 2022-06-03 by rhomboid-shovel

algernon, 2022-06-03:
As the original reporter, you can close the issue. In this case, doing so after you verified that the fix works is perfect.

Thank you!