keyboardio/Chrysalis #863: Layout Card : printing, filtering, and other ideas

Rom1deTroyes, 2022-05-29:
Just coping the feedback from discord (sorry, no time for rephrasing this).

Layout Card (hardly expected feature !) looks great, and the pdf send here by algernon seems promising but :
- Exporting or printing is not obvious : I know Chrysalis is an electron app, so [Ctrl]+[p] must do the job, but it seems to require CUPS, and it is not installed on my system
  - :bulb: Maybe a « Print » button, or something could be more straightforward ?
- Filtering the wanted layers could be something nice : I don't need all my layers because some of them are garbage or testing ones :-)
- When having secondary actions on letters, the main symbol is very small, so peeking those is difficult (not a problem in the editor, but not easy to read as the laminated sheets !

algernon, 2022-05-30:
> Exporting or printing is not obvious : I know Chrysalis is an electron app, so [Ctrl]+[p] must do the job, but it seems to require CUPS, and it is not installed on my system

A print button (which we'd guard with some media queries, so it wouldn't print itself) is something we could add. However, we still rely on the embedded Chromium's printing features, and as such, we will still require whatever that requires.

Out of curiosity, what happens when you press `Ctrl+p`? Does the print dialog not pop up at all? Or does it display a warning? Or does it just silently fail?

> Filtering the wanted layers could be something nice : I don't need all my layers because some of them are garbage or testing ones :-)

Not sure if it's doable, but if we could arrange the layers one per page, you could choose which pages to print from the print dialog. Not sure if that's easier, or adding a filtering system ourselves.

Rom1deTroyes, 2022-06-15:
> Out of curiosity, what happens when you press `Ctrl+p`?
> Does the print dialog not pop up at all? Or does it display a warning? Or does it just silently fail?

Nothing seems to happen : no popup or menu, nothing in the developer console, but 2 lines printed on stdout :

```sh
[81002:0615/231727.992555:ERROR:print_backend_cups.cc(218)] CUPS: Error getting default printer: Mauvais descripteur de fichier
[81002:0615/231727.992587:ERROR:electron_api_web_contents.cc(447)] Failed to get default printer name
```
Just tested the appimage from github, and I get the same error

If I use chromium, the print dialog works, and it can export to pdf.

I can dig around, but my laptop install is very odd (and old), so it may be something like electron using a old unpurged cups config file, or even weirder :-)