keyboardio/Chrysalis #876: Figure out a naming policy for IPC methods

algernon, 2022-05-30:
We don't really have a policy on naming our IPC methods, so we ended up with a bunch of different styles:

- `service.method`: Like `backup.list-library`, `backup.load-file`, etc.
- `service-method`: Like `usb-scan-for-devices`, `focus-find-serial-devices`, etc
- none of the above: `show-devtools`, `devtools-is-open`

Thankfully, we don't have many IPC methods yet, so we can switch to a common policy with little effort. We just need to figure out how we want to name things.

algernon, 2022-05-31:
Additionally to this, we also want a policy - possibly the same one - to cover when to use `ipcMain.on` and `ipcRenderer.send`/`sendSync`, and when to use `ipcMain.handle` and `ipcRenderer.invoke`.

I already see a few places that use `sendSync` when `invoke` would be a better choice.

algernon, 2022-06-03:
To get this rolling, I'd prefer either `service/method` or `service.method`, where `method` would be a verb, most of the time. Thus our current ipc names would become:

- `backups.list-library`, `backups.load-file`, `backups.save-file`
- `usb.scan-for-devices`, `usb.is-device-connected, `usb.find-serial-devices`
- `dialog.file-save-show`, `dialog.file-open-show`
- `devtools.set-state` (or `devtools.open` + `devtools.close`), `devtools.is-open`
- `logging.get-filename`
- `native-theme.should-use-dark-colors`

obra, 2022-06-03:
sure

commit 2626d9f5 (2022-06-04): Harmonize our IPC call naming

This harmonizes our IPC call naming, so everything follows the same pattern.

As an additional benefit, I found an issue with the save/open handlers of
`ipc_file_io`, where errors weren't properly sent back to the renderer, so we
silently hid save errors, among other things. We no longer do that.

Fixes #876.
PR #903 (2022-06-04): Harmonize our IPC call naming
This harmonizes our IPC call naming, so everything follows the same pattern.

As an additional benefit, I found an issue with the save/open handlers of `ipc_file_io`, where errors weren't properly sent back to the renderer, so we silently hid save errors, among other things. We no longer do that.

Fixes #876.
closed 2022-06-04 by obra