keyboardio/Chrysalis #880: "My Keyboard" is getting crowded, again

algernon, 2022-05-31:
The "My Keyboard" tab of Preferences is getting a bit crowded, and while sections are neatly separated, some of the knobs that belong together, aren't. This isn't really a problem _yet_, because most regions only have a single setting, or settings that are stand-alone. But #879 introduces a pair of settings that are bound together, and that makes the tab a lot less clear. We also have a lot of empty screen real estate. On top of all that, the spacing between the elements isn't exactly clear or consistent.

So I've been thinking: what if we re-introduced Cards here? Every setting that belongs together, would be in a card. Or at least on an elevated Paper. Not necessarily stacked below each other, but perhaps in a Grid, or just a flex Box, or a Masonry (from MUI's Labs)?

We should group them up, one way or the other, is what I'm trying to say, really. And not waste horizontal space, either.

Time permitting, I'll post a few mockups to show what I'm thinking of.

obra, 2022-05-31:
Grouping settings is great. I think that the most straightforward solution to the problem you're running into right now is that "LED Preferences" is too general a grouping. (As will be 'Plugin preferences') More smaller sections that live on Papers so they are better broken apart seems like a simple way to deal.

I know that Masonry would be wrong. 

I'm not convinced that wasting horizontal space is a horrible sin.  

I can't think of another desktop application that sticks settings in a grid and looks reasonable.

Just surveying things that are currently open on my Mac:

MS Edge (Material, chrome-derived)

[image]

Excel:
[image]

Slack:
[image]

VS Code:

[image]

Discord:
[image]

WeChat:
[image]

Safari:

[image]

Warp:
[image]

algernon, 2022-06-01:
Progress report:

[image]
[image]

closed 2022-06-02 by obra