keyboardio/Chrysalis #881: Less verbose logging of keyboard probing

obra, 2022-05-31:
If the results of a probe haven't changed, don't log anything. Otherwise, any time on the home screen results in a very crowded log.

algernon, 2022-06-01:
The problem is that we don't keep the previous logs (or results) around, so we can't compare. I think a different solution would be to just route these messages elsewhere.

Since we want to improve our logging story anyway, might aswell hit two birds with the same stone.

It's worth looking at [winston](https://github.com/winstonjs/winston), which seems like a fine logging library, but [pino](https://github.com/pinojs/pino) could be another. Winston probably fits our needs better, though.

algernon, 2022-06-02:
I think I came up with a reasonable idea, using Winston. The nice thing about Winston is that we can create loggers with different combination of formatters, all using the same transport. We don't have to keep the logger instances around, either, because `winston.loggers.get()` will let us retrieve the one we need.

So, with that in mind, we can do the following: in the main process, determine the logfile to use for the session. We use a different file for each session so we can easily slurp it up for the debug bundle. This also has the benefit that we no longer keep the session's log in memory, reducing our footprint. We expose an IPC to let us grab the filename from the renderer process. While we can write to files from the renderer, due to having `nodeIntegration` enabled, we do not have access to `app.getPath("logs")`, so while we could compute the filename in the renderer, it's easier if we do it in the main process. Eventually, we'll want to move all logging to the main process, I suppose, but probably not for 1.0 yet.

But I digress.

So, we have a filename, great. In the renderer, we set up logging ASAP. We'll need to have a `winston.transports.File` instance accessible throughout Chrysalis, so that we can all log to the same file via a shared handle, as to not to corrupt it. I suppose we could stick it in `globalContext`, or wrap it in a singleton or something. Implementation detail.

For the most part, Chrysalis would just use `winston.log()`. For focus, we'd set up a different logger, one that would _not_ echo debug and verbose level logs back to the dev console, only into the file, and would add a `focus` label too - at least on prod builds. In dev builds, we'd still hide debug logs from the console, but would let verbose through. This'd obsolete the verbose logging option in Preferences (which was quite useless to begin with).

We could also have a separate logger for flashing, so we can label those messages appropriately, too.

And perhaps another for toasts... or just use the default for that, since we generally want to see notifications along the logs, and don't really care whether it's a notification or a log message, they're part of the natural flow of logs, so to say.

With this setup, we'll no longer need: `@api/log`, `@api/focus`'s `.debug` and `.debugLog`. We will, however, need to be more conscious about what we log, at what level.

commit fd9197f9 (2022-06-03): Overhaul how Chrysalis does logging

We've been using a home-grown, pretty ad-hoc logging setup, which suffered from
a number of issues. It was simply bad, really. Log levels were all over the
place, logging was inconsistent, sometimes we logged to the console directly,
sometimes we only popped up a toast without logging it, and so on and so forth.

This is an overhaul of the whole thing.

We pull in the Winston logging library to help us out. This lets us store logs
in files too, along with logging to the developer console. We'll use this file
during the debug bundle creation, instead of keep [... cut]
commit 01db9f21 (2022-06-03): Overhaul how Chrysalis does logging

We've been using a home-grown, pretty ad-hoc logging setup, which suffered from
a number of issues. It was simply bad, really. Log levels were all over the
place, logging was inconsistent, sometimes we logged to the console directly,
sometimes we only popped up a toast without logging it, and so on and so forth.

This is an overhaul of the whole thing.

We pull in the Winston logging library to help us out. This lets us store logs
in files too, along with logging to the developer console. We'll use this file
during the debug bundle creation, instead of keep [... cut]
PR #897 (2022-06-03): Overhaul how Chrysalis does logging
We've been using a home-grown, pretty ad-hoc logging setup, which suffered from a number of issues. It was simply bad, really. Log levels were all over the place, logging was inconsistent, sometimes we logged to the console directly, sometimes we only popped up a toast without logging it, and so on and so forth.

This is an overhaul of the whole thing.

We pull in the Winston logging library to help us out. This lets us store logs in files too, along with logging to the developer console. We'll use this file during the debug bundle creation, instead of keepi [... cut]
closed 2022-06-03 by obra