keyboardio/Chrysalis #886: Auto update

obra, 2022-06-02:
For 1.0 we should support autoupdate. These days, I believe it’s largely just something we need to enable.

algernon, 2022-06-02:
Looks like the server-side requirements are quite simple, and Squirrel (the thing that handles auto-updates) has a GitHub Update Manager, so we can just use github releases pretty much as-is.

We may need to switch away from electron-webpack first, so #769 might need to be tackled before we can enable auto-updates. But that's just a hunch.

obra, 2022-06-02:
In theory, we can move away from electron-webpack bit by bit.

We can, I believe, move the packaging toolchain targets without also moving
the dev server and webpack configs.

algernon, 2022-06-05:
It looks like we'll be needing code signing on Windows too, to make this work, at least if we don't want to deploy our own auto update server. And even if we'd have our own update server, implementing auto-update without signing would still be a pain in the backside - mostly for our users who'd have to jump through a number of hoops.

We'll also need to create .zip builds for macOS, and tweak some windows stuff to make the process smoother there. The actual auto-update parts are easy. It's all the infrastructure and build parts that are harder (and more expensive...).

algernon, 2022-06-05:
On the other hand, we are using electron-builder, which has its own update mechanism, and that requires dmg, appimage and an nsis-based exe - things we have, I think. Not sure about nsis, but that's the electron-builder default, so we probably have it.

algernon, 2022-06-05:
Looking at electron-builder's docs, it looks like we'll need to adjust our github action workflows a bit, and rely more on electron-builder to publish. I do not yet see how we'll handle the snapshot builds in this setting, though, I suppose, we don't want auto-update for those anyway, so we can continue doing them as we do now. Thus, we'd only overhaul the release workflow.

algernon, 2022-06-09:
Looks like we won't need to change much in our workflows, either. We'll just need to upload the `latest-*.yml` files to the github releases too. Having played with the auto-update workflow a bit, I have Opinions™.

At least on Linux, there are two ways the update can happen: if the AppImage filename has a version, the updater will download the new one, place it in the same dir as the original, and delete the old. I find this behaviour understandable, but mighty annoying, and - while still understandable - surprising. The other way is when the AppImage didn't have a version, in which case it gets replaced.

I usually have an aversion towards packages that forcibly update themselves, so I think auto-update should be opt-in. If enabled, Chrysalis will notify about new versions, download & install the update, and offer a restart. When not enabled, Chrysalis would only notify about a new version, and link to the release page.

algernon, 2022-06-09:
I think I'm making reasonable progress, things are _mostly_ working, though I've done only minimal testing so far. What my current implementation does not have is a way to cancel a download. That's something I want to add at some point, but it is turning out to be a bit too complex for an initial implementation.

It looks like we could also display the changelog of the update, but that's only really useful if we can ask Chrysalis to not install the update automatically. That's another thing I'd like to do: implement three ways to handle upgrades: manual, download-only, and automatic. Manual would just display a notification if an update is found; download-only would download, and offer a way to view the changelog, which would further offer the choice to upgrade or not; automatic would download & install automatically, no changelog shown.

PR #935 (2022-06-24): Implement an auto-update mechanism
This replaces the old, fragile, latest version notification on the menubar with an auto-upgrade mechanism.

While it is the main process that does the update, it is the renderer that initiates it. We do this so that we have better control over the process, and can display notifications on the renderer, without race conditions.

The renderer orchestrates the process via the `useAutoUpdate` hook, called by the `App`. The hook itself will first tell the main process to check for updates, which in turn will result in an `update-available` event being fired, if th [... cut]
closed 2022-06-24 by obra