keyboardio/Chrysalis #902: Connection Timeout to Model 100 on Win10

rhomboid-shovel, 2022-06-04:
**Describe the bug**
Communication timeout when trying to connect to Model 100

**To Reproduce**
1. Attempt to connect to Model 100
2. Watch the wheel spin
3. See error

**Desktop (please complete the following information):**
 - OS: Win10 21H2 Build 19044.1706
 - Chrysalis Version: Chrysalis 0.10.1-snapshot.266

**Additional context**
This preview worked the first time I connected, I made one small layout change, and then I tried to connect a few minutes later. Can't connect. I tried rebooting the system, didn't help. I tried another Win10 computer, didn't help. 

[2020-06-03 chrysalis-debug.bundle.json.zip](https://github.com/keyboardio/Chrysalis/files/8836297/2020-06-03.chrysalis-debug.bundle.json.zip)

algernon, 2022-06-04:
This is definitely weird. From the logs it is clear that there's ample communication between the Model100 and Chrysalis. The keyboard even responds to a `keymap.default` command, but times out on the `keymap.custom` that follows it immediately after.

I have a few suspicions, but will need to reproduce it locally to dive deeper. Working on that now.

rhomboid-shovel, 2022-06-04:
I should have noted the keymap change I made: changed Layer 0 numlock to be a hyphen. Maybe there's some issue now that Layer 1 is inaccessible? Numlock was the only thing that could get you to Layer 1.

algernon, 2022-06-06:
It looks like we (well, folk much smarter than me) found the issue, and it's not with Chrysalis - it's in the firmware. So we'll need to update to a newer firmware, but we need to connect with Chrysalis first to be able to do that. I'll see what I can do to make that easier.

It does indeed seem to matter that you changed Layer 0 numlock to hyphen. I'll see if I can repro with a similar change - so far I haven't made that change, because it seemed irrelevant.

rhomboid-shovel, 2022-06-06:
Sounds like good progress. Just let me know what to do. I don't urgently
need to use Chrysalis, but I would like to make a couple keymap changes.

algernon, 2022-06-06:
We'll be releasing Chrysalis 0.10.1, with updated firmware that we hope fixes the issue. Once that is out, I'll let you know, and we'll write down instructions about how to get the new firmware onto the keyboard.

algernon, 2022-06-08:
I just released 0.10.1, with a set of firmware that we hope will fix the issue. To flash it, unplug your Model 100, start holding the `PROG` key, and plug it back in while still holding the key. That will put the keyboard into programmable mode, and Chrysalis will be able to flash the update firmware onto it.

rhomboid-shovel, 2022-06-09:
[chrysalis-debug.bundle.json.zip](https://github.com/keyboardio/Chrysalis/files/8869309/chrysalis-debug.bundle.json.zip)

After flashing failure and then a connection attempt

algernon, 2022-06-09:
Progress!

The tool we're using under the hood to flash is crashing on Windows. I'm looking into finding a working version.

algernon, 2022-06-09:
I just pushed an update to master that fixes the tool crash for me, and lets me flash under win10. Build `-snapshot.293` should have the fix, once it is done building in ~20 minutes.

closed 2022-06-09 by rhomboid-shovel