keyboardio/Chrysalis #933: Number of Layers for official firmwares

Rom1deTroyes, 2022-06-09:
**Describe the bug**
In the Atreus experimental firmware, there is 5 layers (to make room for Dynamic Macros in the eeprom).
The default Atreus firmware has 10 layers.

Flashing the experimental firmware just drop the higher layers (hopefully, they are in the backup file)

**To Reproduce**
Steps to reproduce the behavior:
1. Flash Atreus default firmware
2. Put some keys on layers 9 8 7 6 5... and save to keyboard
3. Flash the experimental Atreus firmware
4. See that there are just 5 layers

**Expected behavior**
- A warning if I may loose layers (« Your actual keymaps has 6 layers, the new firmware has 5, you may loose some data : continue anyway ? »
- Having the same number of layers on each official firmware, but the risk is still here when flashing an old custom firmware.

**Desktop (please complete the following information):**
 - OS: Debian GNU/Linux
 - Chrysalis Version: 0.10.2 build from master

algernon, 2022-06-19:
I just finished rearranging the firmware we'll ship in the next Chrysalis release, that will address this issue aswell.

algernon, 2022-06-24:
The updated firmware is merged into master, the number of layers in the Atreus sketch is back up to 10.

closed 2022-06-24 by algernon