keyboardio/Chrysalis #934: Dynamic Macros : Macros are not empty

Rom1deTroyes, 2022-06-09:
**Describe the bug**
When using the Dynamic Macros for the first time on my Atreus, there are some bulk macros already there.

**Expected behavior**
An empty list, or even better : an example macro resuming what you can do :memo: 

**Screenshots**
[image] [image] [image]

**Desktop (please complete the following information):**
 - OS: Debian GNU/Linux
 - Chrysalis Version: 0.10.2-snapshot build from master
 
**Additional context**
I had a custom firmware build from kaleidoscope example sketch that used the eeprom to store 10 layers, and flashed the experimental firmware from Chrysalis.

algernon, 2022-06-09:
This happens when we reuse eeprom space, when you had a firmware that used the area now reserved for dynamic macros for something else, then this happens. The proper fix here would be to clear EEPROM before doing a structured restore, as that will guarantee that we will not see garbage.

I used to be vary - and still am, to some extent - about clearing EEPROM on flash, but clearing + structural restore is the least surprising behaviour, so lets do that, and clear.

algernon, 2022-06-26:
I have a patch that fixes this, but before we can apply that, we need to fortify the flashing process quite a bit. That'll likely take a bit of time, but it's one of my highest priorities now (for a number of reasons, not just this).

algernon, 2022-06-27:
For reference, this depends on #941, we need to make the flashing process more reliable before it is safe to clear EEPROM as part of it.

commit d362785b (2022-06-26): api/flash: Clear the EEPROM after backing it up

In case we flash firmware that changes the Storage layout, we do not want to end
up in a situation where an area that was previously used by one plugin, gets
assigned to another, and have random data in it.

To avoid this situation, we have to clear EEPROM during flashing, right after
backing it up.

Fixes #934.
commit 2f560c6c (2022-07-24): api/flash: Clear EEPROM before flashing

When flashing new firmware, the storage layout may change, and parts of it may
change owners. While we restore the previous contents in a structured manner, if
previously used space changes owner to a plugin that is new in the updated
firmware, it would start with random garbage in its storage slice.

To avoid this situation, clear the EEPROM before flashing, so when we restore,
we restore onto a fresh, empty state.

Fixes #934.
commit aba90e7a (2022-07-26): api/flash: Clear EEPROM before flashing

When flashing new firmware, the storage layout may change, and parts of it may
change owners. While we restore the previous contents in a structured manner, if
previously used space changes owner to a plugin that is new in the updated
firmware, it would start with random garbage in its storage slice.

To avoid this situation, clear the EEPROM before flashing, so when we restore,
we restore onto a fresh, empty state.

Fixes #934.
commit a2e492c2 (2022-07-26): api/flash: Clear EEPROM before flashing

When flashing new firmware, the storage layout may change, and parts of it may
change owners. While we restore the previous contents in a structured manner, if
previously used space changes owner to a plugin that is new in the updated
firmware, it would start with random garbage in its storage slice.

To avoid this situation, clear the EEPROM before flashing, so when we restore,
we restore onto a fresh, empty state.

Fixes #934.
closed 2022-07-26 by algernon