keyboardio/Chrysalis #939: Layout "LOAD FROM FILE..." button hangs Chrysalis

acdingman, 2022-06-09:
**Describe the bug**
"LOAD FROM FILE ..." appears to hang Chrysalis

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Layout & Colormap Editor'
2. Click on 'BACKUP & RESTORE'
3. Click on 'LOAD FROM FILE...'
4. UI becomes unresponsive. No file selection dialog appears. On Mac OS, "Chrysalis -> Quit Chrysalis" may or may not work. On Fedora 36 Gnome "Chrysalis -> Quit" appears to work consistently. No other interaction with the app appears to do anything on either platform

**Expected behavior**
I expected a file selection dialog box to appear, from which I could select a previously exported layout.

**Desktop (please complete the following information):**
 - OS: 
 - - Mac OS 12.4 on an Intel 2018 MBP
 - - Fedora 36 on a 2018 HP Spectre x360
 - Chrysalis Version: 10.1

**Additional context**
I have only tested with one keyboard, a Model 01-Q, serial #000877. After work I can attempt to re-produce with my other Model 01.

acdingman, 2022-06-09:
I've just confirmed that this works fine using the 0.10.0 AppImage on Fedora, so the break is between 0.10.0 and 0.10.1

algernon, 2022-06-09:
Ooops. Thanks for catching it, I reproduced, fix incoming.

commit 04f9d541 (2022-06-09): LayoutSharing: Fix importing from a file

When I renamed all the IPC method names, I forgot to update one. Unfortunately,
it was also a `sendSync` call, so it hung the renderer indefinitely. Updated the
method name now.

Fixes #939.
closed 2022-06-09 by algernon