keyboardio/Chrysalis #941: Ideas to improve the flashing process

algernon, 2022-06-09:
There are a few shortcomings in our current flashing process, one of which is the part where we wait for the keyboard to come back after successful flashing. In general, waiting for the device to come back one way or the other, is a bit fragile. However, the waiting for the bootloader step is easier in practice, because the bootloader triggering key can be pressed way before we get to the stage where we reboot.

On the other hand, after flashing, waiting for the keyboard to boot into the new firmware is more problematic. When the firmware is small, and the flashing is fast, it's possible to have a fairly short (3-7 sec) window to release the trigger key, and that's too short. If it isn't released during that window, the keyboard will remain in bootloader, and we'll never try to get it out of that again, and will ultimately fail the flashing process.

What we should do instead, is look for the bootloader too, and if we found the bootloader, warn the user to release the trigger key, and then reboot the keyboard and wait until we have it booted up, or we time out again. This gives the user another ~20 seconds to release the key.

On top of this, if we still fail the process, we should present a more useful error message, based on the step we failed at. This way if we fail at the reconnect step, we can tell the user that, and offer hints (like releasing the trigger key earlier).

algernon, 2022-06-13:
I was pondering about this over the weekend, and I think I have an idea. Ideally, we want to give ample time to release the trigger key, but at the same time, we don't want to wait longer than needed, either. So I'd propose we change the notifications a bit: after saving EEPROM, instead of immediately triggering the bootloader, we'd pop up a notification that we'll be doing so, and urge the user to press the trigger key *then*. We'd give them a nice ~30 sec timeout. As soon as the bootloader is triggered, we'll turn the notification green, and let them know they can release, and wait another few seconds.

This way we'll have their attention, due to the early "please press now" notification, and they'll only need to pay attention for a little while, a few seconds.

However, the release window still feels short, and I'm not sure how to fix that. If the keyboard is in bootloader mode, can we hit it with another serial HUP to boot back to firmware? What about the Model100 where we don't use the serial HUP to begin with?

The only thing we can do there, is to see if we have the bootloader back, and just wait more. The bootloader should go back to the firmware a bit after releasing the trigger key, so we could wait for that timeout times two or something along those lines.

BrunoChevalier, 2022-06-14:
Is there some kind of graph of the states / state machine used during flashing?
It might help to clarify possible improvements and can be used as documentation.

algernon, 2022-06-15:
There isn't really a state machine, we always go forward, never back, and it is all driven by code written for the particular keyboard.

For example, the Model100's steps are [here](https://github.com/keyboardio/Chrysalis/blob/5891dfa1cf697b6926979eb3ba2978c86623c579/src/api/hardware-keyboardio-model01/index.js#L124-L131), and [this function](https://github.com/keyboardio/Chrysalis/blob/5891dfa1cf697b6926979eb3ba2978c86623c579/src/api/flash/index.js#L233-L272) drives the flashing process.

algernon, 2022-06-26:
I've been mucking around in the flashing code today, cleaned up a lot of stuff, so it's in a state now where I can tackle this issue too.

So, the major problem we want to solve is being reliant on timers. We can control most of the process fine, can wait as long as we need to, roll back a step and restart if need be. The part we have the least control over is the reboot after the flash. For every flasher but Avr109, we can do the reboot explicitly, whenever we want to. For Avr109, it doesn't matter much, because if need be, we can tickle the keyboard with an 1200 hup and force out a reboot.

However, for all this to work, to be able to better discover when we need to reboot, I'll need to rework a few more things. But it sounds doable in a reasonable timeframe.

algernon, 2022-07-10:
Regarding reworking a few things: the explicit reboot after flash thing is something I've been trying to figure out lately, and I just had an idea: `flash()` could return a device-specific function that'd do the reboot when called. This fits in nicely with the existing architecture of `@api/flash`, and doesn't need many changes, but still allows us explicit control over timing the reboot.

algernon, 2022-07-23:
A small update:

https://user-images.githubusercontent.com/17243/180623029-817487eb-20f9-4198-b600-cb025a50e446.mp4

The process is a bit more forgiving now, and does much more useful retries. Changes are around the places where we reboot: from normal to bootloader, and from bootloader to normal.

When going either way, we now retry indefinitely, and pop up a dialog after a few tries to explain the situation. Going normal to bootloader, every iteration we'll try to reboot too, so if someone did not press the trigger key soon enough, they will be able to. Similarly, going bootloader to normal, if we find a bootloader, we'll try to tell it to go into normal mode, and then try again.

My current branch is up at [api/flash/reboot-guard](https://github.com/keyboardio/Chrysalis/tree/api/flash/reboot-guard). So far, only the AVR109 flasher has been updated, the rest will error out. I'll update those too shortly.

closed 2022-07-26 by algernon

BrunoChevalier, 2022-07-27:
Thanks for all the detailed information you gave while updating the code and tackling this issue!

obra, 2022-10-11:
That seems reasonable, generally.

Since we *can* detect whether a board appears or disappears or switches to
or from bootloader mode, we can make this a lot more bulletproof.

To reboot the m100 from bootloader to non-bootloader dfu-util -r *should*
do it.

We don't even have to give them the 30 second timeout. We could just
attempt to reboot the keyboard every 5 seconds until it takes.
(alternatively, for all Keyboardio boards, we *could* have a
device.reboot-to-bootloader focus command that errors out if the interlock
key isn't currently pressed.

commit 862764f4 (2022-07-24): Overhaul of the flashing process

Our old flashing process suffered from a few shortcomings, biggest of which was
its reliance on timing, and retries being woefully incomplete. In practice, this
meant that if the key to help trigger bootloader mode was not pressed soon
enough, or was released too late, we ended up in a state Chrysalis could not
recover from, because it never tried the reset again.

The new process does away with the separate "bootloaderTrigger" and
"bootloaderWait" steps, and combines them into a single "bootloader" step, which
will take care of retrying the reboot too, whethe [... cut]
commit 0b2e1cf5 (2022-07-24): Overhaul of the flashing process

Our old flashing process suffered from a few shortcomings, biggest of which was
its reliance on timing, and retries being woefully incomplete. In practice, this
meant that if the key to help trigger bootloader mode was not pressed soon
enough, or was released too late, we ended up in a state Chrysalis could not
recover from, because it never tried the reset again.

The new process does away with the separate "bootloaderTrigger" and
"bootloaderWait" steps, and combines them into a single "bootloader" step, which
will take care of retrying the reboot too, whethe [... cut]
commit c1071ac3 (2022-07-26): Overhaul of the flashing process

Our old flashing process suffered from a few shortcomings, biggest of which was
its reliance on timing, and retries being woefully incomplete. In practice, this
meant that if the key to help trigger bootloader mode was not pressed soon
enough, or was released too late, we ended up in a state Chrysalis could not
recover from, because it never tried the reset again.

The new process does away with the separate "bootloaderTrigger" and
"bootloaderWait" steps, and combines them into a single "bootloader" step, which
will take care of retrying the reboot too, whethe [... cut]
commit 782f2db9 (2022-07-26): Overhaul of the flashing process

Our old flashing process suffered from a few shortcomings, biggest of which was
its reliance on timing, and retries being woefully incomplete. In practice, this
meant that if the key to help trigger bootloader mode was not pressed soon
enough, or was released too late, we ended up in a state Chrysalis could not
recover from, because it never tried the reset again.

The new process does away with the separate "bootloaderTrigger" and
"bootloaderWait" steps, and combines them into a single "bootloader" step, which
will take care of retrying the reboot too, whethe [... cut]