keyboardio/Chrysalis #949: Keyboard not working after flashing if default layer is not layer 0

urac, 2022-06-16:
**Describe the bug**
My Model01 stops working after flashing firmware

**To Reproduce**
Steps to reproduce the behavior:
1. Backup layout
2. flash newest firmware (I reset to defaults, but that might be irrelevant)
3. load layout file and import
4. upload layout to keyboard

**Expected behavior**
Works as before firmware update

**Desktop (please complete the following information):**
 - OS: Arch Linux
 - Chrysalis Version: 0.10.2

**Additional context**
My layer 0 is empty. That's why my default layer is layer 1.
It seems to default to layer 0 after firmware update, although the chrysalis gui still shows default layer 1 as before the update.
A workaround fix is setting default layer to 0 and then to 1 again.

obra, 2022-06-16:
After you flash your keyboard, can you go into Chrysalis' sidebar menu, click "Report a problem" and generate a debugging bundle and attach it here?

urac, 2022-06-17:
[chrysalis-debug.bundle.zip](https://github.com/keyboardio/Chrysalis/files/8923563/chrysalis-debug.bundle.zip)

bundle created after loading layout from file and saving to keyboard. Keyboard non-working.

algernon, 2022-06-17:
After reviewing the logs, I believe this is a firmware issue, that it doesn't set the appropriate layer on boot. I'll have a closer look later today.

algernon, 2022-06-17:
Yep, confirmed, this is a Kaleidoscope issue. We store the default layer in EEPROM, EEPROMSettings can return it, but it is not switched to during `onSetup()`. We simply need a `Layer.move(settings_.default_layer)` in EEPROMSetting's `onSetup()`.

algernon, 2022-06-17:
Hm, it is a bit more complicated than that, because when `EEPROMSettings::onSetup()` is called, we do not yet know how many layers we have. Calling `Layer.move()` there will work when the default layer is within the bounds of the built-in layers, but it working is purely accidental.

Ideally, we'd have a separate plugin to handle all that, one that can be scheduled to run _after_ stuff like `EEPROMKeymap`'s `::onSetup()`. That, or `Layer.move(EEPROMSettings.default_layer())` in the firmware's `setup()`, called after `Kaleidoscope.setup()`. The latter is less code, and as such, is more practical to do.

So I'd update the firmware shipped with Chrysalis to do just that. Meanwhile, if you're not against compiling slightly modified firmware, you can add the above line to the Model01 firmware, and it should correctly move to layer 1 on boot.

algernon, 2022-06-24:
This is now fixed by the firmware on master, which calls `Layer.move(EEPROMSettings.default_layer())` in `setup()`.

closed 2022-06-24 by algernon