keyboardio/Chrysalis #951: Crashes on >= 0.9.2 when connecting Avermedia BU113

JonathanTroyer, 2022-06-18:
**Describe the bug**
Chrysalis will crash on versions starting in 0.9.2 if an [Avermedia Cap 4k (BU113)](https://www.avermedia.com/us/product-detail/bu113) is connected.

**To Reproduce**
Steps to reproduce the behavior:
1. Launch Chrysalis
2. Connect the device
3. Chrysalis crashes

OR

1. Connect the device
2. Launch Chrysalis
3. Chrysalis crashes as soon as a window is created

**Expected behavior**
Chrysalis does not crash when unrelated USB devices are created

**Screenshots**
NA

**Desktop (please complete the following information):**
 - OS: Windows 11 21H2, build 22000.739
 - Chrysalis Version: any >= 0.9.2

**Additional context**
I can mail my USB device if needed for testing. I'm using an [MSI Z690 motherboard](https://www.msi.com/Motherboard/MEG-Z690I-UNIFY).

JonathanTroyer, 2022-06-27:
Yup! Just point me to it.

algernon, 2022-06-28:
> Yup! Just point me to it.

https://gist.github.com/algernon/f6a2c18d1e6737f713d639b5e8a7b410

Just needs node & the usb module.

As far as I understood, Chrysalis crashes in the main process, so I suspect it's something in the usb detection parts that trips us over. If not, my next suspicion is our device probing, but that currently being in the renderer shouldn't hard-crash Chrysalis, but show an Error screen instead. Nevertheless, if this example does not crash, its output would still be useful, and we can iterate from there.

JonathanTroyer, 2022-06-28:
Oh boy.

If I have the BU113 connected, the main loop never runs (I get to "Starting main loop...", but never "Scanning for devices..."). Stepping in with a debugger, the function itself never gets called before node exits.

If I have it disconnected, it'll run with this output:
```js
{
  devices: [
    {
      busNumber: 1,
      deviceAddress: 2,
      deviceDescriptor: [Object],
      portNumbers: [Array]
    },
    {
      busNumber: 1,
      deviceAddress: 1,
      deviceDescriptor: [Object],
      portNumbers: [Array],
      interfaces: undefined
    },
    {
      busNumber: 1,
      deviceAddress: 6,
      deviceDescriptor: [Object],
      portNumbers: [Array]
    },
    { busNumber: 2, deviceAddress: 0, deviceDescriptor: [Object] },
    {
      busNumber: 1,
      deviceAddress: 4,
      deviceDescriptor: [Object],
      portNumbers: [Array],
      interfaces: undefined
    },
    {
      busNumber: 1,
      deviceAddress: 5,
      deviceDescriptor: [Object],
      portNumbers: [Array],
      interfaces: undefined
    },
    {
      busNumber: 1,
      deviceAddress: 3,
      deviceDescriptor: [Object],
      portNumbers: [Array],
      interfaces: undefined
    },
    {
      busNumber: 1,
      deviceAddress: 8,
      deviceDescriptor: [Object],
      portNumbers: [Array]
    },
    {
      busNumber: 1,
      deviceAddress: 7,
      deviceDescriptor: [Object],
      portNumbers: [Array],
      interfaces: undefined
    },
    { busNumber: 1, deviceAddress: 0, deviceDescriptor: [Object] }
  ]
}
```

If I plug in the BU113 while it's running, the event loop will crash when it next runs. Not on `getDeviceList()` like I thought, but on the actual next execution of the event loop.

algernon, 2022-06-29:
> If I plug in the BU113 while the loop is running, the event loop will crash when it next runs. Not on `getDeviceList()` like I thought, but on the actual next execution of the event loop.

That's weird. Hm. Can you try the script again, after removing the `addUsbEventListeners();` line?

algernon, 2022-06-29:
Mhm, it looks like the problem is somewhere in the notifications then. I think we'll have to take it up with the node-usb people, I'll do that soon.

obra, 2022-10-11:
Hi Jonathan,

Does this happen with other USB devices connected or only with the
Avermedia device connected?

JonathanTroyer, 2022-10-13:
As far as I can tell, it only happens with the Avermedia device

obra, 2024-02-26:
Chrysalis now runs in-browser using WebSerial and WebUSB, so I'm closing out this issue as obsolete. Please don't hesitate to open a new issue if https://chrysalis.keyboard.io exhibits the same behavior

closed 2024-02-26 by obra