keyboardio/Chrysalis #954: Command and Control Flipped on MacOS

James-Firth, 2022-06-21:
**Describe the bug**
When I set a key on my Keyboardio Atreus to be the Left `Command` key (or as a modifier) I get `Control` behaviour and vice-versa

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Chrysalis with an Atreus on MacOS
2. Set a key as the left `Command` key
3. Save
4. Open a browser
5. Press "Command" and Tab
6. You'll get the Control + Tab behaviour

**Expected behavior**
Command on the Atreus should behave the same as Command on a standard Macbook Keyboard.

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: MacOS Monterey
 - Chrysalis Version: 0.10.2
 
 This issue happened previously but Chrysalis was new enough I assumed it was a small bug and would get fixed, apologies for not reporting sooner.

**Additional context**
N/A

obra, 2022-06-22:
I’m unable to repro this. Have you previously switched those keys on the Atreus using the macos keyboard preferences settings?

James-Firth, 2022-06-22:
@obra Shoot. Yup that was it, I feel silly now. 🤦 Thanks for the help!

I'm a relatively new Mac user and must have flipped it so my Atreus would work between my machines when I had a different desk setup a year ago and completely forgot.

Apologies for the extra notification!

[image]

closed 2022-06-22 by James-Firth