keyboardio/Chrysalis #955: Improve the name & wording of "Secondary Actions"

algernon, 2022-06-21:
See the discussion on Discord starting [here](https://discord.com/channels/492408953041321984/801534424726503435/988907739898077215).

But the gist of it is that "Secondary action" in itself isn't very clear. As suggested during the discussion, a name with "held" in it would likely be clearer, and "Action when held" sounds like a better name than "Secondary Actions".

gergelypolonkai, 2022-06-22:
I like the idea of having a small question mark next to the text; in that case you don’t really have to change the wording (thus save on screen estate) just provide a good explanation.

Nabi Yang’s help text is a good start.

```
Have this key perform a different action when held (for example, Ctrl). If you want to have the original action repeat, just double tap and hold the key.

This feature is achieved using the Qukeys plugin.
```

I added the Qukeys line so people who are familiar with Qukeys but not with Chrysalis can still have an idea what this feature does. That line definitely needs some improvement, but the idea could be used in other places (like the Sticky modifier keys and “Layer Shift for next action” (OneShot)

chrisSCM, 2022-11-29:
"Sticky Action"?