keyboardio/Chrysalis #963: Some of the Preferences widgets don't adapt well to smaller window sizes

algernon, 2022-06-25:
[image]

Screenshot says it all.

algernon, 2022-06-28:
It's not only Preferences, but a couple of other places too. The Changelog looks silly at 800x600 too, among other things.

obra, 2022-06-28:
mui has some layout-width dependent properties we can use for stuff like this.

algernon, 2022-06-28:
> mui has some layout-width dependent properties we can use for stuff like this.

Yep, but for these, we didn't need to, see #969. We could still use those layout-width dependent props to improve Preferences further (reduce the right margin in the tab on smaller window sizes), but that's much lower on my prio list than fixing actual layout bugs.