keyboardio/Chrysalis #965: Updating keyboard firmware broke my keyboard (NEED URGENT HELP)

necsoiu, 2022-06-26:
Trying to update Atreus firmware I receive an update error, and after that the keyboard was disconnected from Chrysalis. Trying to reconect Chrysalis says "**Reading data from device...**" and after staying in this state from a while it says "**Communication timeout**".

So my keyboard is now useless. Please advice me what should I do to recover my keyboard functionality.

I also want to point out that there is too much inconsistency in the firmware update process. While here (https://shop.keyboard.io/pages/atreus-getting-started) it says to hold down the Esc key until the end in fact in the update process somewhere in step 3 we are told that if we have any key pressed it should be released. Which variant is correct?

Please also tell me where can I see the firmware version installed on a keyboard.

algernon, 2022-06-26:
> Trying to update Atreus firmware I receive an update error, and after that the keyboard was disconnected from Chrysalis. Trying to reconect Chrysalis says "**Reading data from device...**" and after staying in this state from a while it says "**Communication timeout**".

If  you're running a recent (0.10.x) version of Chrysalis, it saves *very* useful logs. On Linux, they'll be in `~/.config/Chrysalis/logs`, on macOS, under `~/Library/Application Support/Chrysalis/logs`, and on Windows, `%APPDATA%/Chrysalis/logs`.

Can you upload the latest logfile, please?

For future reference, there's a "Report a problem" menu item in Chrysalis, which lets you save the current logs to a file to a custom location, where it's easier to upload them from (and the debug bundle saved from there includes additional information useful for us for debugging, like the operating system used, etc).

> So my keyboard is now useless. Please advice me what should I do to recover my keyboard functionality.

If you held the Esc key during the entire process, then simply unplugging & replugging the keyboard should help. It is likely stuck in bootloader mode.

If that does not help, then the most straightforward step would be to unplug your keyboard, hold the `Esc` key (or rather, the key in the bottom left corner), and plug it in while the key is being held. That'll put the keyboard into programmable mode, and Chrysalis will be able to flash new firmware onto it.

This should restore functionality.

> I also want to point out that there is too much inconsistency in the firmware update process. While here (https://shop.keyboard.io/pages/atreus-getting-started) it says to hold down the Esc key [...]

Chrysalis is correct. Esc should be held until the flashing process itself starts, and released before it ends. Chrysalis performs some additional steps before and after the flashing itself, such as saving & restoring customisations stored in EEPROM.

We're working on making the flashing process smoother, and less confusing, but that's still a work in progress.

> Please also tell me where can I see the firmware version installed on a keyboard.

At the moment, there isn't a way. That's also something we're working on, but it isn't ready yet.

necsoiu, 2022-06-26:
> f you held the Esc key during the entire process, then simply unplugging & replugging the keyboard should help. It is likely stuck in bootloader mode.
>
> If that does not help, then the most [...]

None of the above worked. 

As observation:
- The other keyboard is seen in Widows Devices as Atreus while the broken one is seen as USB Serial device (COM4).
- Unpluging the keyboard, removing USB Serial device (COM4) from Devices and then pluging again give me the following windows messages:
     a) Setting up a device (We're setting up 'Atreus Bootloader'. and then
     b) Device is ready ('USB Serial Device (COM4) is set up and ready to go.
After theat as I mentioned in Windows Devices I have a USB Serial device (COM4) device.

However although COM4 is visible in Chrysalis 0.10.2 it cannot connect to this keyboard.

What else could I try?

algernon, 2022-06-26:
We have resolved the situation over on Discord. I'll be opening a few issues later today, based on the logs received.

closed 2022-06-26 by algernon