keyboardio/Chrysalis #967: 962 breaks device discovery on macos.

obra, 2022-06-28:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem. If possible, please also include the debug bundle you can create on the "System Information" screen. If, for some reason, you are unable to access that screen, please press `Control + Shift + I` to bring up the developer console. Right-clicking inside it will allow you to save its contents to a file - please attach that to the bug report.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Chrysalis Version: [e.g. 0.4.0]

**Additional context**
Add any other context about the problem here.

obra, 2022-06-28:
I'm going to revert 962 temporarily

algernon, 2022-06-28:
Ooops. Problem is us calling `ipcRenderer.sendSync("udev.isAvailable")` unconditionally. I'll revert the revert and fix that.

Sorry!

commit 51b751d3 (2022-06-28): Reapply #962, with non-linux platforms fixed

We need `/run/udev` to be able to scan devices, so if it is not accessible, or
doesn't exist, we will now display a useful error message. As this is Linux
specific, make sure that the `udev.isAvailable` IPC endpoint works on the other
platforms too, in case it is called by accident, and also don't invoke the IPC
unless we're on Linux.

This fixes #967, and reintroduces the fix for #700.
closed 2022-06-28 by algernon