keyboardio/Chrysalis #980: Update the Dutch translation

algernon, 2022-07-01:
I plan to cut a new Chrysalis release sometime next week, it would be nice if the Dutch translation could be updated for it.

@BrunoChevalier, @anderium, would either of you be up for the task?

We have moved the translation stuff to [Weblate](https://hosted.weblate.org/projects/chrysalis/master/nl/), and I think that updating the translation there would be easier than updating the `nl.json` file directly.

If anything is unclear, and needs context, let me know, and I'll add a screenshot to Weblate to show where and how a particular string is used, or add some context, and so on.

BrunoChevalier, 2022-07-01:
Sure, I'll take a look tonight or later this weekend.

algernon, 2022-07-01:
Fixed in #979, thank you!

closed 2022-07-01 by algernon